zephyrproject-rtos/zephyr #15012: Unable to establish security after reconnect to dongle

pdunaj, 2019-03-29:
**Describe the bug**
We have two devices: mouse (peripheral) and dongle (central). Dongle connects to mouse when the other one is available. Until around two weeks ago everything was working fine but lately security cannot be established after one of the device reconnects.

I found out that the problem is in `ll_enc_req_send` that rejects starting an encryption
```
#if defined(CONFIG_BT_CTLR_PHY)
        if ((conn->llcp_req != conn->llcp_ack) ||
            (conn->llcp_phy.req != conn->llcp_phy.ack)) {
#else /* CONFIG_BT_CTLR_PHY */
        if (conn->llcp_req != conn->llcp_ack) {
#endif /* CONFIG_BT_CTLR_PHY */
                LOG_ERR("PDUNAJ security: %s %d", __func__, __LINE__);
                return BT_HCI_ERR_CMD_DISALLOWED;
```

There was recently a change in this area that was adding check for a collision (`f4c8e3ff3197d36773d68b4b4802daabc401f918`)

**To Reproduce**
Bond peripheral and central. Request security level medium from peripheral. Note security established. Disconnect or reset one of the devices. Reconnect. Observe no error on security request. Observe that no security is established.

**Expected behavior**
When security is requested we should get an error or security should be established.

**Impact**
Showstopper

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
ncs zephyr rev: `de69d2df908fc7bccd8c03b106e7645e90b1503b` (upstream: `373a42b`)

**Additional context**
N/A

carlescufi, 2019-03-29:
Setting high priority since this is a critical regression

pdunaj, 2019-04-04:
Do anybody knows if this will be fixed? If fix will not happen soon, can the change causing a problem be reverted?

carlescufi, 2019-04-04:
@pdunaj yes, @joerchan is working on this and it's his first priority.

PR #15186 (2019-04-10): Bluetooth: controller: Set encryption procedure pending
Set the encryption procedure pending and start it as soon as
there are no other procedures running.
This allows the Host to enqueue the encryption procedure, and not
get disallowed command because a different LL procedure is active.
Fixes #15012

Stop the Length request procedure from initiating during the encryption
procedue. This would cause the peer to disconnect the link with error
code LMP_TRANSACTION_COLLISION

Send unknown rsp instead of asserting when encryption related control
PDUs are received in the wrong state.
This would allo [... cut]
commit 0cf82cdd (2019-04-10): Bluetooth: controller: Set encryption procedure pending

Set the encryption procedure pending and start it as soon as
there are no other procedures running.
This allows the Host to enqueue the encryption procedure, and not
get disallowed command because a different LL procedure is active.

Fixes #15012
PR #15335 (2019-04-11): Bluetooth: controller: Fix encryption setup to be queueable 
Fix the controller implementation to make start encryption
queueable if there is any control procedure in progress.

The context related to encryption procedure is not shared so
that it will be used as the ongoing procedure completes.

The fix here maintains the old functionality of serializing
the queued data and LL Encryption Request PDU, so that data
queued before start encryption is acknowledged.

Fixes #15012. (which was marked as fixed but still had an outstanding issue)
commit fff35bd6 (2019-04-11): Bluetooth: controller: Fix encryption setup to be queueable

Fix the controller implementation to make start encryption
queueable if there is any control procedure in progress.

The context related to encryption procedure is now shared so
that it will be used after the ongoing procedure completes.

The fix here maintains the old functionality of serializing
the queued data and LL Encryption Request PDU, so that data
queued before start encryption is acknowledged.

Fixes #15012.
closed 2019-04-10 by carlescufi