zephyrproject-rtos/zephyr #15013: tests/kernel/fatal: check_stack_overflow: (rv equal to TC_FAIL)

cinlyooi-intel, 2019-03-29:
**Describe the bug**
This is a different bug from @13934 because it occurred at a different line

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_se_c1000_ss_devboard -DARCH=arc
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-2-gf5a4a698ec17 (delayed boot 1000ms) *****
Running test suite fatal
===================================================================
starting test - test_fatal
test alt thread 1: generic CPU exception
Exception vector: 0x8, cause code: 0x0, parameter 0x0
Address 0x400004d8
Current thread ID = 0xa8000400
Faulting instruction address = 0x400004d8
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
***** Kernel OOPS! *****
Current thread ID = 0xa8000400
Caught system error -- reason 4
test alt thread 3: initiate kernel panic
***** Kernel Panic! *****
Current thread ID = 0xa8000400
Caught system error -- reason 5
test stack sentinel overflow - timer irq
posting 1024 bytes of junk to stack...
busy waiting...
FAIL - stack_sentinel_timer@147. should never see this
***** Stack Check Fail! *****
Current thread ID = 0xa8000400
Caught system error -- reason 2

    Assertion failed at /home/cinlyooi/tests/zephyr.git/tests/kernel/fatal/src/main.c:193: check_stack_overflow: (rv equal to TC_FAIL)
thread was not aborted
FAIL - test_fatal
===================================================================
Test suite fatal failed.
===================================================================
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain zephyr
 - Commit SHA or Version used: f5a4a698ec17e6a78299966a1a3fe73c3404b78d

**Additional context**
Add any other context about the problem here.

aescolar, 2019-03-29:
@cinlyooi-intel, thanks for the report, to speed up things it would be a bit nicer, please:
* refer to bugs using GH notation #XYZ so they are automatically linked
* label the bug with the appropriate area (if you can)

Note that commit you mention cannot be found in the upstream tree (is there a copy paste mistake?)

pizi-nordic, 2019-03-29:
I run this test using current master (d3bb3cfd7a4a2358ba7927d5913ecbeb5cfaf577) on the nrf51_pca10028, nrf52_pca10040 and nrf52840_pca10056. The test was repeated 100 times on each board. All iterations passed successfully on all boards.

Could somebody run this on the quark_se_c1000_ss_devboard?

nashif, 2019-03-29:
@pizi-nordic it fails for me on nrf52840_pca10056.
Note that there are two tests, the test that fails is 

`tests/kernel/fatal/kernel.common.stack_sentinel`

pizi-nordic, 2019-03-29:
@nashif: Ehhh. I used prj.conf. Allow me to check that again...

pizi-nordic, 2019-03-29:
@nashif: I have re-run all the tests using sentinel.conf in the same manner as described here: https://github.com/zephyrproject-rtos/zephyr/issues/15013#issuecomment-477986903

I have not seen even single occurrence of the bug.

andrewboie, 2019-03-29:
I've seen this from time to time in CI runs, I've definitely seen it on qemu_xtensa and mps2_an385.

andrewboie, 2019-03-29:
```
test stack sentinel overflow - timer irq
posting 1024 bytes of junk to stack...
busy waiting...
FAIL - stack_sentinel_timer@147. should never see this
```

This is the same bug as #13934

andrewboie, 2019-03-29:
This is really a problem with the test case. On a tickless system, you can busy wait as long as you want but you're not guaranteed to get a timer IRQ. I need to contemplate a better way to exercise this. Meanwhile, downgrading to 'low'.

andrewboie, 2019-03-29:
PR sent, putting back to medium as this causes spurious CI failures in some cases even with retries

nashif, 2019-03-30:
> @nashif: I have re-run all the tests using sentinel.conf in the same manner as described here: [#15013 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/15013#issuecomment-477986903). I have not seen even single occurrence of the bug.

ok, looks like this happens with CONFIG_BOOT_DELAY=1000 in prj.conf, something the hardware test setup does...

closed 2019-03-30 by nashif

nashif, 2019-03-30:
With fix from @andrewboie this is now resolved also with `CONFIG_BOOT_DELAY=1000`

PR #15032 (2019-03-30): tests: fatal: fix sentinel timer IRQ checking
Tickless kernel is now always disabled, ensuring that when
the kernel's tick count changes, we really did get a timer
interrupt.

The test now awaits a change in tick count instead of busy
waiting for an arbitrary time period.

We move the stack sentinel check in z_get_next_switch_handle()
to before the code that determines what the next thread to run
should be.

Fixes: #15013
Fixes: #15037
commit dea4394e (2019-03-30): tests: fatal: fix sentinel timer IRQ checking

Tickless kernel is now always disabled, ensuring that when
the kernel's tick count changes, we really did get a timer
interrupt.

The test now awaits a change in tick count instead of busy
waiting for an arbitrary time period.

Fixes: #15013