zephyrproject-rtos/zephyr #15017: Not able to set "0xFFFF No specific value" for GAP PPCP structured data

akshhr, 2019-03-29:
Currently, it is not possible to set the following parameters to "0xFFFF No specific value" 
CONFIG_BT_PERIPHERAL_PREF_MIN_INT
CONFIG_BT_PERIPHERAL_PREF_MAX_INT
CONFIG_BT_PERIPHERAL_PREF_TIMEOUT
This is required as per the specification document, core v5.0, vol 3, part C, section 12.3 Peripheral preferred connection parameters characteristic.

jhedberg, 2019-03-29:
So the range needs to be extended to 0xffff in Kconfig and then we probably also want a `BUILD_ASSERT(within_range(val) || val == 0xffff)`

Vudentz, 2019-04-02:
But if one don't want to specify the value why to select it in the first place? I guess it would have been better not to include the attributes if you won't specify any value.

jhedberg, 2019-04-02:
> But if one don't want to specify the value why to select it in the first place? I guess it would have been better not to include the attributes if you won't specify any value.

@Vudentz maybe because you want a specific value on one or two of the parameters, but don't care about the value of the rest?

akshhr, 2019-04-02:
@Vudentz I agree with @jhedberg and also it is a requirement from the specification.

PR #15117 (2019-04-02): Bluetooth: PPCP: Fix not being able to disable configurations
The spec allows to disable certain configurations with use of 0xffff,
to restrict the range that can be entered the value 0 is used instead.

As this still enables entering values in the invalid range, 1-5 for
min/max interval, this adds the necessary build checks to prevents
values within this range to be used and at same time check if min
interval is not bigger that max interval.

Fixes #15017
commit 593d8561 (2019-04-02): Bluetooth: PPCP: Fix not being able to disable configurations

The spec allows to set to no specific value with use of 0xffff.

As this still enables entering values in the invalid range, 3200-65535
for min/max interval, this adds the necessary build checks to prevent
values within this range to be used and at same time check if min
interval is not bigger that max interval.

Fixes #15017
closed 2019-04-02 by nashif