zephyrproject-rtos/zephyr #15021: set the RTC time & date manually and getting timestamps with ms/ns resolution

StefJar, 2019-03-29:
To sync distributed systems, I like to set the RTC time and date.
It would be perfect if the RTC API provide an API for that.

Maybe getting a (date)timestamp directly out of the API would be useful as well.

So far the ZephrOS documentation is not clearly telling, from which source the OS is generating its ticks and how it behaves when we have more than one possible source available.
Using the time functions of the ANSI C lib would be the "best" way. But they are lacking  ms/ns resolution.

StefJar, 2019-04-25:
I wrote some basic code for the stm32f4 platform. I uploaded a blueprint for that:
https://github.com/StefJar/zephyr_stm32_rtc_driver

JulienR59, 2022-06-01:
How do you set the time of the RTC without entering init mode first?

StefJar, 2022-08-29:
the example code is a hack

https://github.com/StefJar/zephyr_stm32_rtc_driver/blob/master/rtc_stm32.c#L155

makes the zephyr build system to run the init function at the kernel startup before the main function is called