zephyrproject-rtos/zephyr #15027: doc: PDF generation broken

dbkinder, 2019-03-29:
**Describe the bug**
Generating a PDF version of the Zephyr documentation displays some errors and creates PDF output with bad formatting in the Network Stack Architecture section through the end of that chapter.  Note that this documentation section uses a floating image to create a two-column effect that renders fine for the HTML docs, but apparently not for the PDF.

[image]

**To Reproduce**
Do a ``make pdfdocs``

**Impact**
We currently don't have a requirement to create a PDF version of the documentation, but such a requirement is likely needed in the future.

jukkar, 2019-04-01:
Even thou this issue is seen with networking documentation, I think it is more about an issue in documentation system itself, so assigning to David for a time being.

dbkinder, 2019-11-11:
PDF build failure resolved with #19255

closed 2019-11-11 by dbkinder