zephyrproject-rtos/zephyr #15031: net: 9cd547f53b "Fix ref counting for the net_pkt" allegedly broke reference counting

pfalcon, 2019-03-29:
Culprit, based on dissecting: 9cd547f53b7474945e94111e5f8ce9b05fa9579c

Steps to reproduce: samples/net/sockets/dumb_http_server BOARD=qemu_x86 OVERLAY=overlay-e1000.conf (which got added only in 946836577bb1adfe62e25771b8d565d2074d162f), then as usual run `ab -n1000 http://192.0.2.1:8080/`.

Seen behavior (100% and exactly reproducible):

~~~
Benchmarking 192.0.2.1 (be patient)
apr_pollset_poll: The timeout specified has expired (70007)
Total of 14 requests completed
~~~

Expected behavior: 1000 requests should go thru.

pfalcon, 2019-03-29:
This would be "Priority: High". But as I'm not a release manager, I don't set it.

pfalcon, 2019-03-29:
Also reproducible with default set up of dumb_http_server (i.e. qemu_x86 + SLIP).

pfalcon, 2019-03-29:
Also reproducible with BOARD=mps2_an385 CONF_FILE="prj.conf overlay-smsc911x.conf".

jukkar, 2019-03-29:
Hmm, I will investigate this. The original commit I sent fixed a real issue. It seems that the fix exposed some other issue.

PR #15045 (2019-04-01): Fix TCP refcount issue
Marking this as high as it impacts typical TCP use cases and fixes memory leak.

Fixes: #15031
Fixes: #15050
commit 30d31936 (2019-04-01): Revert "net: tcp: Fix ref counting for the net_pkt"

This reverts commit 9cd547f53b7474945e94111e5f8ce9b05fa9579c.

The commit we are reverting, fixed originally the issue that was
seen with zperf. There we freed the net_pkt too early while it was
still waiting for a TCP ACK. The commit 9cd547f5 seemd to fix that
issue but it was causing issues in dump_http_server sample app which
then started to leak memory. No issues were seen with echo-server
with or without the commit 9cd547f5.

So the lessons learned here is that one needs to test with multiple
network sample apps like dump_http_server, e [... cut]
closed 2019-04-01 by nashif