zephyrproject-rtos/zephyr #15044: ARC: test failure in tests/kernel/threads/thread_apis

andrewboie, 2019-03-30:
Seen on nsim_sem_mpu_stack_guard and nsim_em_mpu_stack_guard

```
starting test - test_k_thread_foreach
TEST: Newly added thread found
TEST: tid: 0x80000204, prio: 0

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/kernel/threads/thread_apis/src/test_kthread_for_each.c:90: test_k_thread_foreach: ((tcount > count) && thread_flag && stack_flag is false)
thread_callback() not getting called
FAIL - test_k_thread_foreach

```

vonhust, 2019-04-01:
@andrewboie @ruuddw   this is caused by 
`if ((char *)thread->stack_info.start ==
			K_THREAD_STACK_BUFFER(tstack)) {`

one of issues caused by the mismatch in stack_info.  It's not a critical issue. I plan to clear them in 1.15
When MPU_STACK_GUARD is enabled. The layout of thread's stack area will be

| User stack layout |
| ------ |
| user stack defind in K_THREAD_STACK_DEFINE  |
| stack guard |
| fixed size privilege stack |

and 

| Kernel stack layout |
| ------ |
| stack guard |
| defind in K_THREAD_STACK_DEFINE |
| fixed size privilege stack (which can be merged with the up part) |

for user thread `stack_info.start == K_THREAD_STACK_BUFFER(tstack)`
for kernel thread `stack_info.start == K_THREAD_STACK_BUFFER(tstack) + STACK_GUARD_SIZE`

When MPU_STACK_GUARD is not enabled, it's no proble,

So as the arch will set `stack_info` according to the type of thread,  it's dynamic, should not rely on 
static info like  K_THREAD_STACK_BUFFER.

If in 1.15,  other archs like ARM plan to refer arc's stack allocation mechnism , there will be similar  issues.

andrewboie, 2019-04-02:
the problem is that the test is trying to do too much. PR sent.

PR #15122 (2019-04-03): tests: thread_apis: fix failure on ARC
This test is only trying to prove that k_thread_foreach() works,
it has nothing to do with stacks. Remove the stack checks
completely.

Fixes: #15044
commit 468c47bb (2019-04-03): tests: thread_apis: fix failure on ARC

This test is only trying to prove that k_thread_foreach() works,
it has nothing to do with stacks. Remove the stack checks
completely.

Fixes: #15044
closed 2019-04-03 by nashif