zephyrproject-rtos/zephyr #15046: Native_posix command line handling

jukkar, 2019-03-30:
I just compiled `samples/net/sockets/dump_http_server` which does not enable UART by default. I then tried to start the sample app using this command `zephyr/zephyr.exe -attach_uart` but got an error `Incorrect option '-attach_uart'`. It took some time to figure out what was going on.

Anyway, my point was that the all the command line options should be always available. If the option is not really usable with that configuration, it should/could tell the user that "option is not available with this configuration" and possibly give some hint what to enable in zephyr config file.

aescolar, 2019-03-30:
( Changing to enhancement, as this is indeed an enhancement request, not a bug :) )

Uhm... The thing is, many options exist only when a given driver/component/application is compiled in. There is no central place where all possible options are known.

It would be certainly possible to change the error message, and help message, so they would hint the user about why an option is not available (Say: "Incorrect option %s. Did you miss-spell it? Is that feature supported in this build?" )
@jukkar : Would that be an acceptable enough solution?

jukkar, 2019-03-30:
After looking how the command line options are done, I understand that is indeed difficult task to do what I was proposing. Your suggestion for a hint "Incorrect option %s. Did you miss-spell it? Is that feature supported in this build?"  sounds like a good workaround.

PR #15048 (2019-04-18): native_posix: cmdline: Hint user about wrong options
Be a bit more friendly to users, by providing some hints
about possible reasons why a command line option was not
understood.
Also describe in the help message that which options are
avaliable depends on what has been selected in this build.

Fixes: #15046
commit 149cc7fd (2019-04-18): native_posix: cmdline: Hint user about wrong options

Be a bit more friendly to users, by providing some hints
about possible reasons why a command line option was not
understood.
Also describe in the help message that which options are
avaliable depends on what has been selected in this build.

Fixes: #15046
closed 2019-04-18 by nashif