zephyrproject-rtos/zephyr #15063: tests/subsys/settings/fcb/src/settings_test_save_unaligned.c fail with assertion failure on nrf52_pca10040

cinlyooi-intel, 2019-04-01:
Exactly the same as #13734. Created new bug instead of reopening because of the time passed since bug was closed.

The tests in tests/subsys/settings/fcb/src/settings_test_save_unaligned.c fail with assertion failure on nrf52_pca10040

**Commit ID: ** 89735fb401a82ac64414d14331733ab1db5ce3f

**Arch:** arm
**Platforms:** nrf52_pca10040

**Error on console log:**
nrf52_pca10040:arm
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-506-g406e942949 (delayed boot 1000ms) *****
===================================================================
starting test - test_config_save_fcb_unaligned

Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/subsys/settings/fcb/src/settings_test_save_unaligned.c:40: test_config_save_fcb_unaligned: (val8_un == 33 is false)
bad value read
FAIL - test_config_save_fcb_unaligned
===================================================================
Test suite test_config_fcb failed.
PROJECT EXECUTION FAILED
```

nvlsianpu, 2019-04-04:
@cinlyooi-intel 

Didn't you  run this test without waiting for execution end before failure? This indeed cause failure which is consequence of unexpected flash area content.

nvlsianpu, 2019-04-04:
This is testing suite bug (test suite need to be fixed) - not setting subsystem bug.

PR #15189 (2019-04-08): tests/subsys/settings/fcb: fix unaligned test
Unaligned test might filed in case fcb area was not clen before
run.
The patch insert clean operation before this test.

fixes #15063
commit bfb7e453 (2019-04-08): tests/subsys/settings/fcb: fix unaligned test

Unaligned test might filed in case fcb area was not clen before
run.
The patch insert clean operation before this test.

fixes #15063
closed 2019-04-08 by nashif