zephyrproject-rtos/zephyr #15065: [Question] How to override/disable compiler flags added by Zephyr

ivakyb, 2019-04-01:
These flags are added when compiling with `west build -b stm32f4_disco .`:
```
-Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -nostdinc -g -Wall 
-Wformat -Wformat-security -Wno-format-zero-length 
-imacros home/kyb/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main 
-fno-common -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables 
-fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable 
-fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith 
-ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -MD -MT 
CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles/app.dir/src/main.cpp.obj.d 
-o CMakeFiles/app.dir/src/main.cpp.obj -c ../src/main.cpp
```

Flags set in project root `CMakeLists.txt`

    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17")
    set(CMAKE_CXX_FLAGS  "${CMAKE_CXX_FLAGS} -Wall -Wextra -Wno-unused-variable -Wno-comment")

appear before flags added by Zephyr.

I also tried 

    set_target_properties( app PROPERTIES COMPILE_FLAGS "-Wno-unused-variable -Wno-comment" )

Zephyr flags still "win".

#### I need at least disable `-nostdinc` and set `-std=c++17` and also may be others.

### How to override flags added by Zephyr?

*<sub>Thanks for your attention</sub>*

gon1332, 2019-04-01:
@nikooikono

SebastianBoe, 2019-04-03:
For which C files do you need to override the flags? Just the app, or all files?

SebastianBoe, 2019-04-03:
Try

zephyr_library_compile_options(-std=c++17 -Wall -Wextra -Wno-unused-variable -Wno-comment)

from the app CMakeListst.txt after boilerplate.cmake.

ivakyb, 2019-04-03:
@SebastianBoe  I need to override GCC flags only for files of my project, no sense to touch zephyr's files.

`zephyr_library_compile_options` works fine. Thanks.

closed 2019-04-03 by ivakyb