zephyrproject-rtos/zephyr #15073: Device crashes when starting with USB connected

pdunaj, 2019-04-01:
**Describe the bug**
```
ASSERTION FAIL [m_drv_state == NRFX_DRV_STATE_INITIALIZED] @ /home/pdunaj/work/ncs/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1708
```

```
void nrfx_usbd_enable(void)                                                                                              
{                                                                                                                        
    NRFX_ASSERT(m_drv_state == NRFX_DRV_STATE_INITIALIZED);
```

**To Reproduce**
Steps to reproduce the behavior:
Connect USB cable, observe USB working fine. Reset device - observe device dying.

**Expected behavior**
Device should be able to work fine when starting with USB connected.

**Impact**
Showstopper

**Screenshots or console output**

**Environment (please complete the following information):**
zephyr ncs: `0bf5263b0522bb5cfac84eefdfdee86dc2c67e3e` (upstream `d3bb3cf`)

**Additional context**
N/A

pdunaj, 2019-04-01:
Hi @pawelzadrozniak , is this something new?

pdunaj, 2019-04-01:
```
[00000045] <inf> event_manager: e: module_state_event module:main state:READY
[00000145] <inf> hid_state: Init HID state!
[00000146] <dbg> usb_nrfx.usb_dc_ep_check_cap: ep 81, mps 16, type 3
[00000150] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0.
[00000150] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 128.
[00000152] <inf> power_manager: Activate power manager
[00000159] <inf> event_manager: e: module_state_event module:board state:READY
[00000161] <inf> event_manager: e: module_state_event module:buttons state:READY
[00000164] <inf> event_manager: e: module_state_event module:wheel state:READY
[00000167] <inf> event_manager: e: module_state_event module:leds state:READY
[00000169] <inf> event_manager: e: module_state_event module:battery state:READY
[00000172] <inf> event_manager: e: module_state_event module:usb_state state:READY
[00000175] <inf> event_manager: e: led_event led_id:0 mode:ON period:0 color: < 255 0 20 >
[00000180] <inf> event_manager: e: led_event led_id:1 mode:BLINKING period:1000 color: < 150 0 0 >
[00000260] <wrn> bt_hci_core: No ID address. App must call settings_load()
[00000260] <inf> ble_state: Bluetooth initialized
[00000262] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00000263] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00000263] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
ASSERTION FAIL [m_drv_state == NRFX_DRV_STATE_INITIALIZED] @ /home/pdunaj/work/ncs/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1708
***** Kernel Panic! *****
Current thread ID = 0x20004174
Faulting instruction address = 0xbc00
```

carlescufi, 2019-04-01:
@pdunaj can you revert https://github.com/zephyrproject-rtos/zephyr/pull/14875/commits/8604d086669a193e35b63b138a9c0f4baadf06d4 and see if it fixes it?

jfischer-phytec-iot, 2019-04-01:
It is broken after bdde5fd98489013b7ac477e8549f218497bf3f06, https://github.com/zephyrproject-rtos/zephyr/pull/14875.

With current master, the device does not exit the while loop inside [nrfx_usbd_enable()](https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c#L1748)

```
    while (0 == (NRF_USBD_EVENTCAUSE_READY_MASK & nrf_usbd_eventcause_get()))                            
    {                                                                                                    
        /* Empty loop */                                                                                 
    }            
```

pdunaj, 2019-04-01:
Hi @carlescufi , sure. Will be back in a sec.

pdunaj, 2019-04-01:
Hi @carlescufi , I reverted `bdde5fd98489013b7ac477e8549f218497bf3f06` (seems to be the commit you referred to on the ncs fork) and issue is gone. If we cannot do quick revert and upmerge then I propose to revert it on fork until fix is ready upstream.

PR #15114 (2019-04-02): drivers: usb: nordic: Fix for startup with USB cable attached.
Fixes bug introduced in #14875.
USBDETECTED event is be generated on cable attachment and
when cable is already attached during reset, but not when
the peripheral is re-initialized. When USB-enabled bootloader
is used, target application will not receive this event
and it needs to be generated again. This commit implements
a check against driver being enabled multiple times in case
of application without bootlader starting with cable attached
(both "fake" and "real" events are generated). Such dirty
trick allows bootloader to leave [... cut]
commit e01662d3 (2019-04-02): drivers: usb: nordic: Fix for startup with USB cable attached.

Fixes bug introduced in #14875.
USBDETECTED event is be generated on cable attachment and
when cable is already attached during reset, but not when
the peripheral is re-initialized. When USB-enabled bootloader
is used, target application will not receive this event
and it needs to be generated again. This commit implements
a check against driver being enabled multiple times in case
of application without bootlader starting with cable attached
(both "fake" and "real" events are generated). Such dirty
trick allows bootloader to leav [... cut]
closed 2019-04-02 by nashif