zephyrproject-rtos/zephyr #15077: Cannot boot application flashed to nrf52840_pca10059

pdunaj, 2019-04-01:
**Describe the bug**
For some reason Zephyr chooses to flash it at 0x1000 not 0x0 for scenario without bootloaded.

**To Reproduce**
Steps to reproduce the behavior:
Build and flash dongle.

**Expected behavior**
Should boot from 0x0 if no bootloader support enabled.

**Impact**
Showstopper.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
zephyr ncs: `0bf5263b0522bb5cfac84eefdfdee86dc2c67e3e` (upstream `d3bb3cf`)

**Additional context**
N/A

pdunaj, 2019-04-01:
Probably somebody change configuration in DTS yet again...

pdunaj, 2019-04-01:
Probably this one `500fa722fb4ae971f2f0668bc02af9c001168e42`. I guess only one configuration is assumed and other ignored.

pdunaj, 2019-04-01:
Hi @lemrey , we don't use MBR and so would like load address be 0 if no mcuboot and slot0 if mcuboot is present. As it is now we cannot boot our dongle at all as `FLASH_LOAD_OFFSET` points to incorrect place in memory.

pdunaj, 2019-04-01:
The problem is BOARD_HAS_NRF5_BOOTLOADER. Can this be converted into project specific options instead of a board option?

lemrey, 2019-04-01:
`nrf52840_pca10059` has no debugger, so the default behavior is different from other boards; you have to use `nrfutil` to program it. This behavior is documented ([here](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10059/doc/index.html)), as are the instructions to have it instead be linked from 0x00.

Can you achieve the desired behavior by selecting the `fstab-debugger` partition file in the board DTS file?

pdunaj, 2019-04-02:
Hi @lemrey , we have jlink connector mounted on this board. There are pads for it so such action is in line with board design.  We can access board using a debugger. Why limiting board use to a specific configuration only when there is absolutely no need for it?
I understand forcing boot step can be convenient for some group of users who are not willing to mount a jlink connector. But this also has serious drawbacks: it complicates process (yet another tool to use), increases bootup time (additional step) and it uses flash. So it would be great if it would be optional.

I can force the behavior by setting `CONFIG_BOARD_HAS_NRF5_BOOTLOADER=n` but it would be better if this variable would be called `CONFIG_NRF5_BOOTLOADER_PERSENT` as it would imply project/system feature rather than board's.

lemrey, 2019-04-02:
Hi @pdunaj, the functionality is not limited to a specific configuration, simply, there is a default configuration. The reasoning behind the default configuration is that the board comes with no debugger and a bootloader, so I thought that Zephyr application for `nrf52840_pca10059` should support that configuration by default.

I don't understand how having `CONFIG_NRF5_BOOTLOADER_PRESENT`, defaulting to `off` would be any different from what we currently have (except for a different default setting). Why setting `CONFIG_BOARD_HAS_NRF5_BOOTLOADER=n` in your project (and selecting the provided `fstab-debugger` partitions), is not an acceptable solution for you?

I understand that you have a different use case than the default one, but if we change the default I am sure in one month there is going to be another issue asking why aren't applications working with the bootloader.

pdunaj, 2019-04-02:
Hi @lemrey , the default is fine and I agree it should stay as is to avoid confusion. We can set the configuration option to disable bootloader.

As for the name, I have impression (maybe it is wrong) that options `HAS_xyz` just like refer to characteristic (compatibility) of HW and so should not be modified by application configuration. If presence of the bootloader is optional having "HAS_" in the name may be misleading. I may be totally wrong here however.

pdunaj, 2019-04-02:
We are setting the `TEXT_SECTION_OFFSET` in our configuration file. So far it was working and it seems that the latest change should not really affect it. Probably the order in which values are applied is different now.

I disabled `BOARD_HAS_NRF5_BOOTLOADER` and dongle boots. Let's leave it as is until somebody else will complain about the name.

closed 2019-04-02 by pdunaj