zephyrproject-rtos/zephyr #15096: Cannot build samples/net/ipv4_autoconf

pfl, 2019-04-01:
Building samples/net/ipv4_autoconf for sam_e70_explained fails with errors. Upstream is at 1eead93a483bbb758e8ad418f3f715b4fe6d285c.
```
$ cd samples/net/ipv4_autoconf
$ make build && cd build
$ cmake -GNinja -DBOARD=sam_e70_xplained ..
$ ninja
```
...
```
zephyr/drivers/ethernet/eth_sam_gmac.c:60:2: error: #error Remaining free RX data buffers (CONFIG_NET_BUF_RX_COUNT -
 #error Remaining free RX data buffers (CONFIG_NET_BUF_RX_COUNT -
  ^~~~~
In file included from <command-line>:
zephyr/samples/net/ipv4_autoconf/build/zephyr/include/generated/autoconf.h:138:42: error: expected identifier or '(' before numeric constant
 #define CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT 12
                                          ^~
zephyr/drivers/ethernet/eth_sam_gmac.c:61:2: note: in expansion of macro 'CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT'
  CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT) * CONFIG_NET_BUF_DATA_SIZE
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/drivers/ethernet/eth_sam_gmac.c:1982:12: error: 'rx_desc_que0' undeclared here (not in a function); did you mean 'tx_desc_que0'?
     .buf = rx_desc_que0,
            ^~~~~~~~~~~~
            tx_desc_que0
[68/143] Building C object zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/exc_manage.c.obj
```

rveerama1, 2019-04-03:
@pfl PR provided and given explanation there. Please take a look.

pfl, 2019-04-03:
Still failing with:
```
zephyr/drivers/ethernet/eth_sam_gmac.c:60:2: error: #error (CONFIG_NET_BUF_RX_COUNT - CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT) * CONFIG_NET_BUF_DATA_SIZE are not large enough to hold a full frame
 #error (CONFIG_NET_BUF_RX_COUNT - CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT) * \
  ^~~~~
[58/150] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/swap.c.obj
```

rveerama1, 2019-04-04:
I already mentioned this in PR.
---------------------------------------------------------------------------------------------------------------------------
But it still gives error when you compile samples/net/ipv4_autoconf sample for sam_e70_xplained board. Its because of low RX buffer count in prj.conf. If someone wants to build this sample please increase the Rx Buffer count.
---------------------------------------------------------------------------------------------------------------------------

pfl, 2019-04-04:
> please increase the Rx Buffer count.

Shouldn't that be "the bug" for sam_e70_xplained in this sample then? Why is the prj.conf left in such a low buffer count state in the first place?

pfl, 2019-04-04:
Can we have a .prj file for sam_e70_xplained to get this sample working out of the box?

rveerama1, 2019-04-05:
@pfl submitted fix.

PR #15143 (2019-04-03): Fix compilation warning
Line continuation was missing for #error macro.

Fixes  #15096

But it still gives error when you compile samples/net/ipv4_autoconf sample for sam_e70_xplained board. Its because of low RX buffer count in prj.conf. If someone wants to build this sample please increase the Rx Buffer count.
commit ecdef392 (2019-04-03): drivers: sam_gmac: Fix compilation warning

Line continuation was missing for #error macro.

Fixes  #15096
PR #15203 (2019-04-05): Increase buffer count to build ipv4_autoconf sample for sam_e70_xplained board
More buffers are need to build for sam_e70_xplained board to work out of the box. Added board specific conf file.

Fixes #15096
commit 06159d3b (2019-04-05): net: samples: ipv4_autoconf: Need more buffers for same_70

More buffers are need to build for sam_e70_xplained board
to work out of the box. Added board specific conf file.

Fixes #15096
closed 2019-04-05 by nashif