zephyrproject-rtos/zephyr #15098: Bluetooth: GATT Unsubscribe doesn't work if a new parameters are used.

paulgavrikov, 2019-04-01:
**Describe the bug**
If `bt_gatt_unsubscribe` is called with other parameters than the subscribe was called it will not work and return `-EINVAL` because it does not actually compare the handles.

**Expected behavior**
Unsubscribe should happen by comparing the handles.

Vudentz, 2019-04-02:
It removes the subscription parameters, it is there is the documentation, if you want to unsubscribe by handle that is a different matter but the ownership of the subscription is still within the params so I like to understand why such API would be needed.

paulgavrikov, 2019-04-03:
@Vudentz Simply, to not have to bother storing the params. Especially with multiple subscriptions.

Vudentz, 2019-04-04:
@paulgavrikov Im not following, one most store the params since it is pushed to a list in the stack in order to call notifify, and yes it does need to be multiple params in order to support multiple notification, the stack don't copy any params.

paulgavrikov, 2019-04-08:
@Vudentz Right, and this works for the Zephyr sample but if you have multiple connections with multiple subscriptions issued by a user to any potential GATT server handling this becomes a nightmare.

Vudentz, 2019-04-09:
@paulgavrikov Most of Zephyr API actually have the same principle of not copying parameters, that seems natural when there is no use of malloc internally. Alternatives would be to have a pool on the stack, but that seems more complicated than having one struct per request which each application should be able to managed compared to managing everything with a single pool which is prone to be misconfigured causing it to fail at runtime.

carlescufi, 2020-03-17:
@Vudentz and @paulgavrikov does this still apply, or should we close it?

closed 2024-07-05 by aescolar