zephyrproject-rtos/zephyr #15100: Bluetooth: GATT (un)subscribe can silently fail

paulgavrikov, 2019-04-01:
**Describe the bug**
`bt_gatt_subscribe` / `bt_gatt_unsubscribe` can fail (e.g. subscription to unknown handle) without notifying the user because the ATT response is ignored.

**Expected behavior**
Add `err` param to callback and trigger on response (similar to GATT write / read) or add a completely new callback just for subscribe responses.

Vudentz, 2019-04-02:
The API is not only about unsubscribing the CCC but also unregistering the parameters and removing it from the list so it can be reused, if we do have to wait the CCC write result that would mean the parameters cannot be reused, ever, if the CCC write fails.

Btw, Im not sure why are you coming up with some rather unusual situation, in this case the handle can only be invalid if it had been changed, in which case all we can do is to invalidate the subscription anyway, actually I don't see any outcome where the CCC write fails and we do want to keep the subscription active thus Im closing this as invalid.

closed 2019-04-02 by Vudentz

paulgavrikov, 2019-04-03:
@Vudentz All other APIs (i.e. read / write) notify the callback on success or fail. What happens if a subscription is issued to a non_existing handle? The other APIs handle that.
With `bt_gatt_subscribe` we have no feedback about that. As I commented on Slack subscribe followed by an unsubscribe before the ATT response crashes the system.