zephyrproject-rtos/zephyr #15103: nrf52810_pca10040 SRAM space not enough

wentongwu, 2019-04-02:
**Describe the bug**
it seems nrf52810_pca10040 SRAM space is not enough for current version.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52810_pca10040  ..
3. ninja
4. See error

**Screenshots or console output**

for case tests/kernel/sched/preempt on platform nrf52810_pca10040
Memory region         Used Size  Region Size  %age Used
           FLASH:       63104 B       192 KB     32.10%
            SRAM:         25 KB        24 KB    104.17%
        IDT_LIST:          56 B         2 KB      2.73/opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr/zephyr_prebuilt.elf section `priv_stacks_noinit' will not fit in region `SRAM'
/opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000020006000,0000000020006037] overlaps section priv_stacks_noinit VMA [0000000020003800,00000000200063ff]
/opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 1024 bytes
collect2: error: ld returned 1 exit status

for case tests/kernel/poll on platform nrf52810_pca10040
-sdk-0.10/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -S zephyr.elf > zephyr.lst && /opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf -e zephyr.elf > zephyr.stat
/opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr/zephyr.elf section `priv_stacks_noinit' will not fit in region `SRAM'
/opt/zephyr-sdk-0.10/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 2048 bytes

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain (Zephyr SDK, 0.10.0)
 - Commit SHA 193fb971c24f827464982307b5b3bb34de0fa98d

cinlyooi-intel, 2019-04-02:
Seeing the same error at 193fb971c24f827464982307b5b3bb34de0fa98d

wentongwu, 2019-04-02:
@andrewboie  this is the same issue when PR#15005 run CI test.

ioannisg, 2019-04-02:
nrf52810 is a low-end SOC - several tests do not run for this device, anyways.

I am wondering whether it is a problem of increasing the PRIVILEGED_STACK_SIZE to default of 1024.

wentongwu, 2019-04-02:
> nrf52810 is a low-end SOC - several tests do not run for this device, anyways.
 I'm not running cases on that platform, but CI test want to do that when I submit PR #15005.  
> I am wondering whether it is a problem of increasing the PRIVILEGED_STACK_SIZE to default of 1024.
you mean megered patches increasing PRIVILEGED_STACK_SIZE cause this platform's failure?

andrewboie, 2019-04-02:
> I am wondering whether it is a problem of increasing the PRIVILEGED_STACK_SIZE to default of 1024.

My guess as well. How many threads does this test create?

wentongwu, 2019-04-04:
> > I am wondering whether it is a problem of increasing the PRIVILEGED_STACK_SIZE to default of 1024.
> 
> My guess as well. How many threads does this test create?

from priv_stacks_hash.c, sched/preempt has 11 threads and /kernel/poll has 10 threads.

wentongwu, 2019-04-04:
for sched/preempt/, if PRIVILEGED_STACK_SIZE = 256, zephyr_prebuilt.elf is 15872 Bytes, so if set PRIVILEGED_STACK_SIZE = 1024, zephyr_prebuilt.elf will almost occupy the whole sram. Finally it should add kobjects, so when do the linking for zephyr.elf, there isn't enough space.

I see the commit changing priv stack size from 256 to 1024, so the work for proving maximum syscall stack depth is ongoing? @andrewboie

andrewboie, 2019-04-05:
@wentongwu I can't reproduce the failure in tests/kernel/sched/preempt, can you retry?
```
[ 90%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       62584 B       192 KB     31.83%
            SRAM:         23 KB        24 KB     95.83%
        IDT_LIST:          56 B         2 KB      2.73%
[ 90%] Built target zephyr_prebuilt
```

PR #15230 (2019-04-06): userspace: reduce CI userspace testing to tests that use it and fix some network tests
- CONFIG_TEST_USERSPACE is now off by default. It should be enabled for any test that tries to put a thread in user mode (K_USER option to k_thread_create()). A special assertion has been added to enforce this. This should *greatly* decrease CI build times.

- Missing system call for gethostname() added.

- The getnameinfo() test case now runs in user mode as intended.

- tests/kernel/poll uses a lot less RAM by recycling stack/thread objects

Fixes: #15228 
Fixes: #15227 
Partial fix for: #8559 (we eventual [... cut]
commit 7b1ee5cf (2019-04-06): tests: CONFIG_TEST_USERSPACE now off by default

Unlike CONFIG_HW_STACK_PROTECTION, which greatly helps
expose stack overflows in test code, activating
userspace without putting threads in user mode is of
very limited value.

Now CONFIG_TEST_USERSPACE is off by default. Any test
which puts threads in user mode will need to set
CONFIG_TEST_USERSPACE.

This should greatly increase sanitycheck build times
as there is non-trivial build time overhead to
enabling this feature. This also allows some tests
which failed the build on RAM-constrained platforms
to compile properly.

tests/drivers/build_al [... cut]
closed 2019-04-06 by nashif