zephyrproject-rtos/zephyr #15107: samples/application_development/code_relocation fails to build with coverage on mps2_an385

dcpleung, 2019-04-02:
**Describe the bug**
When building `samples/application_development/code_relocation` on `mps2_an385` with coverage enabled, it complains that

```
ERROR: section .sram2_data is 6828 bytes in the virtual/runtime address space, but only 6824 in the loaded/XIP section!
```

Note that the size of section `.sram2_data` in `zephyr.map` is 6824 (0x1AA8) bytes:

```
.sram2_data     0x00000000202002d4     0x1aa8 load address 0x0000000000000394
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mps2_an385 -DCONFIG_COVERAGE=y ../
3. make
4. See error

**Expected behavior**
Should build and run without errors.

**Impact**
Impacting code coverage.

nashif, 2019-04-02:
maybe we should just exclude it when coverage is enabled, this is a very "custom" sample

dcpleung, 2019-04-02:
Is there a way to exclude it from coverage in the yaml file?

nashif, 2019-04-03:
no, just add `CONFIG_COVERAGE=n` to the project file.

dcpleung, 2019-04-03:
From what I can tell, `check_link_map.py` doesn't take into account the actual size of the section (in this case 0x1AA8), but tries to infer size with addresses. The section can actually fit inside the relocated (loaded/XIP) space. I don't know enough about relocation and the script to fix it correctly.

AdithyaBaglody, 2019-04-08:
There are 2 options here. 
1. Disable the coverage. 
2. Run this on a hardware which has multiple memory regions.
 The 2nd option is not feasible for mps2_an385 because it has only one memory region. This sample creates another memory region inside sram to show the working of code-relocation script. I am not sure how the gcov would behave when we instrument the code in different memory regions. So I suggest we go with first option and disable coverage for this sample.

wentongwu, 2019-04-08:
.sram2_data     0x00000000202002d4     0x1aa8 load address 0x0000000000000394
                     ... ...
 .data..LPBX1   0x0000000020201d70        0x4 CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
 .data..LPBX1   0x0000000020201d74        0x8 libcode_relocation_source_lib.a(code_relocation.c.obj)
 *(SORT_BY_ALIGNMENT(.rel.data..LPBX1))
                0x0000000020201d7c                . = ALIGN (0x4)
                0x0000000020201d7c                __sram2_data_end = .
                0x00000000202002d4                __sram2_data_start = ADDR (.sram2_data)
                0x0000000000001aa8                __sram2_data_size = SIZEOF (.sram2_data)
                0x0000000000000394                __sram2_data_rom_start = LOADADDR (.sram2_data)

.sram2_bss      0x0000000020201d80       0x98 load address 0x0000000000001e3c

@AdithyaBaglody I have a question: why vma of sram2_bss is  0x0000000020201d80 instead of 0x0000000020201d7c? I think it should be the problem, Thanks a lot!

AdithyaBaglody, 2019-04-08:
I actually don't see this error. I just find that the emulator has crashed. In my case I don't see 4byte offset (but still fails) ,so I cant tell what caused this. But if you think this is the issue please go ahead an create a PR fixing it.

closed 2019-04-09 by nashif