zephyrproject-rtos/zephyr #15109: ATSAME70 MCU(SAM E70 Xplained) RAM random after a watchdog reset.

overheat, 2019-04-02:
**Describe the bug**
After digging deeper with GDB, I found that section .noinit be cleared(random) during the boot process on ATSAME70 MCU (SAM E70 Xplained board), Which make test/driver/watchdog keep resetting.

PS: According to ARM spec, .noinit is used for non-initialized objects that will not be cleared during the boot process.

**To Reproduce**
Steps to reproduce the behavior:
1.west build -b sam_e70_xplained tests/drivers/watchdog/wdt_basic_api/
2.west flash
4. See error

**Expected behavior**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

**Environment (please complete the following information):**
 - Toolchain (e.g Zephyr SDK, ...)
 - latest

ioannisg, 2019-04-02:
FYI @aurel32

aurel32, 2019-04-02:
Yes, it's also what I noticed there: https://github.com/zephyrproject-rtos/zephyr/issues/13417#issuecomment-471099233

I have the impression it comes from the SoC itself and not from Zephyr. The datasheet seems to imply that, at least it doesn't say the RAM is preserved when the watchdog triggers.

nashif, 2019-04-02:
duplicate of #13417

closed 2019-04-02 by nashif

overheat, 2019-04-02:
@aurel32 Yes, I agree with you , it comes from the SoC itself. So should we exclude this chip(ATSAME70) from watchdog test, do you know how to do it?