zephyrproject-rtos/zephyr #15116: nrfx_twim: (nRF52) driver unable to perform i2c_burst_write() correctly

jfischer-phytec-iot, 2019-04-02:
**Describe the bug**

nrfx_twim (nRF52) driver is not able to perform i2c_burst_write(...) correctly, the driver handles the transfer of two messages as two separate transfers with "start repeat" in between. 

Though there is NRFX_TWIM_XFER_TXTX type of transfer and NRFX_TWIM_XFER_DESC_TXTX macro, but also in this configuration a "restart" is present between the transfers, I did not found a way to suppress it.

Below example of two messages in i2c_burst_write() and captured sda+scl waveform:
```
        msg[0].buf = &start_addr;  /* 0x40 */                                                                      
        msg[0].len = 1U;                                                                                 
        msg[0].flags = I2C_MSG_WRITE;                                                                    
                                                                                                         
        msg[1].buf = (u8_t *)buf; /* 0xff    0xff    0xff    0x1f    0xc7    0xf7    0xf7 ... */                                                     
        msg[1].len = num_bytes;  /* 1024 */                                                                         
        msg[1].flags = I2C_MSG_WRITE | I2C_MSG_STOP; 
```

[image]

**To Reproduce**
Steps to reproduce the behavior:
-1. cd zephyr-project/zephyr/samples/display/cfb
0. add reel_board.overlay
```
&i2c0 {
        status = "ok";

        ssd1306@3c {
                compatible = "solomon,ssd1306fb";
                reg = <0x3c>;
                label = "SSD1306";
                width = <128>;
                height = <64>;
                segment-offset = <0>;
                page-offset = <0>;
                display-offset = <0>;
                segment-remap;
                com-invdir;
                prechargep = <0x22>;
        };
};
```
1. mkdir build; cd build
2. cmake -DBOARD=reel_board ../ && make menuconfig #disable SSD1673 and enable SSD1306
3. make
4. connect LA to sda + scl

**Expected behavior**
[image]

**Impact**
i2c display controller can not be used with nRF52 platform

**relevant issues and PR**
https://github.com/zephyrproject-rtos/zephyr/issues/11612
https://github.com/zephyrproject-rtos/zephyr/pull/14015

carlescufi, 2019-05-27:
@Mierunski @anangl could you please comment on this issue?

anangl, 2019-05-27:
This is not a driver issue but the limitation of the hardware. In TWIM peripheral, it is required to provide the buffer to be used and the length of the transfer when starting it, and these two values cannot be later on altered for this particular transfer. Only a new transfer (resulting in generation of the start condition on the bus) can be started with different parameters. This makes it impossible to merge two transfers from different buffers into a single transaction on the bus, as it seems to be expected by `i2c_burst_write` (there is a related [warning](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/i2c.h#L491) in the I2C API documentation). Such merging is only achievable with TWI, although the `nrfx_twi.c` driver does not provide a corresponding option yet (it is planned to add such enhancement).

jfischer-phytec-iot, 2019-05-27:
>(there is a related warning in the I2C API documentation)

This warning was added because TWI(M) does not support it, see 46072632bd2ca23adef6b2bcb4527afe8e55f035

pabigot, 2019-07-27:
This is a duplicate of #11612, see resolution summary [here](https://github.com/zephyrproject-rtos/zephyr/issues/11612#issuecomment-462443654).

PR #17801 (2019-07-29): drivers: i2c_nrfx_twi: Make use of NRFX_TWI_FLAG_SUSPEND
Make use of the new flag introduced in nrfx v1.7.2 that suspends
transfers, in order to properly handle scattered I2C transactions,
especially to perform i2c_burst_write() in the way expected by
display drivers, i.e. as a continuous transaction on the bus,
without the repeated START between its two parts.

Fixes #15116 (although the issue mentions the TWIM peripheral, on which the hardware limitation prevents the issue from being solved, on all SoCs of nRF52 Series TWI can be used instead of TWIM).
closed 2019-07-29 by carlescufi