zephyrproject-rtos/zephyr #15119: GPIO callback not disabled from an interrupt

pdunaj, 2019-04-02:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/issues/10186 is back again.

In new hw version we can no longer use single callback as we have pins scattered on two ports. We need to disable callbacks for both ports at the same time in the single interrupt. Yet the current infrastructure makes it impossible.

I will narrow it to nrf52 case. `gpiote_event_handler` first checks which ports/pins fired and then trigger callbacks for port 0 followed by callbacks for port 1.
In our case we want to disable interrupt for port 1 when servicing port 0 interrupt. The way function is implemented it does not work as local copy of what fired was already made.

**To Reproduce**
Steps to reproduce the behavior:
Configure pin interrupt on two ports. Hook a callback to each port. Disable both callbacks from isr. Trigger interrupt on both pins at the same time. Observed that disable does not work for port 1 and isr is still called.

**Expected behavior**
Disable isr should not be called.

**Impact**
Can be serious if not handled correctly/unexpected.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
ncs zephyr: `0bf5263b0522bb5cfac84eefdfdee86dc2c67e3e` (upstream `d3bb3cf`)

**Additional context**
N/A

pdunaj, 2019-04-02:
There is a way to fix this in the driver. When `gpio_nrfx_pin_disable_callback` and `gpio_nrfx_pin_enable_callback` are called one could remove bit corresponding to pin in `pin_mask` member on all callbacks corresponding with the port (these functions know which port/pin is touched).
Actually one should record the original mask (as set by callback init) to see if bit should really be restored.

ioannisg, 2019-07-19:
@anangl @pdunaj is this issue being looked at?

pdunaj, 2019-07-19:
hi @ioannisg , we have applied a workaround and not really track if this is being changed. This can be a problem for somebody who does not expect it to happen.

ioannisg, 2019-07-19:
> 
> 
> hi @ioannisg , we have applied a workaround and not really track if this is being changed. This can be a problem for somebody who does not expect it to happen.

@pdunaj you've applied a workaround in the application code, I assume?
So the issue still exists, right?

pdunaj, 2019-07-19:
@ioannisg , yes - all changes to make this work are in the application's code.

PR #18214 (2019-08-14): drivers: gpio_nrfx: Correct the way the callbacks are fired
Add additional masking of the pins with fired callback triggers
against the currently enabled callbacks, in order to not call
handlers for callbacks that got disabled in some other callback
handlers that were called in the same ISR execution.

Fixes #15119.
PR #18260 (2019-11-07): [Backport v1.14-branch] drivers: gpio_nrfx: Correct the way the callbacks are fired
Backport #18214.

Fixes #15119.
closed 2019-08-14 by ioannisg