zephyrproject-rtos/zephyr #15133: Is the level2 interrupt supported for ARM cortex-M0P core?

lylezhu2014, 2019-04-03:
For ARM cortex-M0P core, the MAX IRQ Line is 32 for peripherals.  In order to support more, the level2 inetrrupt controller is added.
Is the level2 interrupt supported for ARM cortex-M0P core? If yes, how to use this feature. If no, is there a plan to support this feature?

aescolar, 2019-04-14:
@ioannisg Can you help here?

ioannisg, 2019-04-14:
A bit confused here, are you talking about an external interrupt controller, not NVIC?

lylezhu2014, 2019-05-09:
Yes. Both the external interrupt controller (a perperial of the ARM core) and NVIC exist for the same one core.

mbolivar, 2019-05-12:
@ioannisg @lylezhu2014 Zephyr has generic multi-level interrupt support for all architectures. The driver API is `struct irq_next_level_api`. If the M0+ has another interrupt controller built in at architecture level, I'd start by looking for that API implemented for it. If not, I suppose it would need to be added.

ioannisg, 2019-05-13:
Yeah, the api is not implemented for cortex-m, afaics.

ioannisg, 2019-05-13:
Haven't heard any discussions about supporting this in the stack, do you want to file a feature request @lylezhu2014 ?

lylezhu2014, 2019-05-15:
@ioannisg I created a ticket #16168 to address this requirement.

ioannisg, 2019-05-15:
Great, then, I think you can close this one

closed 2019-05-15 by lylezhu2014