zephyrproject-rtos/zephyr #15144: West: how to set up for external modules

Laczen, 2019-04-03:
Hi, I would like to use west for a multi repository setup. My idea is to setup a zephyr directory as:
```
/zephyrproject/zephyr
/zephyrproject/extmod/mod1
/zephyrproject/extmod/mod2
...
/zephyrproject/extboard/board1
/zephyrproject/extboard/board2
```
where extmod and extboard would have there own repository.

From the documentation it is clear how to setup each of the subdirectories /extmod/mod1, ... however it is not completely clear for me how to reflect this for west. 
Should I add all the modules to the west.yml under zephyr ? Or is is better to remove the west.yml under zephyr and to construct a west.yml under /zephyrproject ?
Is there some kind of setup example ?

carlescufi, 2019-04-03:
Hi @Laczen. Should I assume you are setting up a downstream with your own repositories that are complementary of the upstream zephyr? If so then you are very close to the Nordic usecase, and you can see how we handle that here:

https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/west.yml

Note that we have our own Zephyr fork, but you can point to upstream zephyr in your west.yml if you prefer.

carlescufi, 2019-04-03:
Also see https://docs.zephyrproject.org/latest/guides/west/why.html#topologies-supported. We use Topology 2.

Laczen, 2019-04-03:
@carlescufi, thanks for the reply. I am indeed very close to the Nordic usecase. So basically the idea is to create a new west.yml manifest above the zephyr directory. But how do I get started?
Step 1: I create a west.yml, but what do I do with the self section (especially how to make sure the west commands are still working) ?
Step 2: How to call west init and update for the created west.yml ?

carlescufi, 2019-04-03:
@Laczen 

Step 1. You create a `west.yml` in your main "downstream" repository, let's assume it's called "foo":

`foo/west.yml`:

```
west:
  url: https://github.com/zephyrproject-rtos/west
  revision: master

manifest:
  defaults:
    remote: laczen

  remotes:
    - name: laczen
      url-base: https://github.com/laczen
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      path: zephyr
      west-commands: scripts/west-commands.yml
      remote: zephyrproject
      revision: <whatever revision you want>
   <add projects here>

  self:
    path: foo
```

Step 2.

```
west init -m https://github.com/laczen/foo
west update
```

will give you a set of 2 repositories at the same level, `zephyr` and `foo` and all of the flash commands will still work.

Laczen, 2019-04-03:
@carlescufi , cool, thanks.

closed 2019-04-03 by Laczen

gon1332, 2019-05-24:
@carlescufi I followed your directions, but I'm having trouble on the `west init` part. It is always resulting to:
```
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'post-init' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')
```

I have this [test repo](https://gitlab.com/gon1332/west-test-project) that I want to use as a manifest with the T2 star west topology.

The **west.yaml**:
```{YAML}
west:
  url: https://github.com/zephyrproject-rtos/west
  revision: master

manifest:
  defaults:
    remote: gon1332

  remotes:
    - name: gon1332
      url-base: https://gitlab.com/gon1332
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      path: zephyr
      west-commands: scripts/west-commands.yml
      remote: zephyrproject
      revision: v1.14.0

  self:
    path: west-test-project
```

The **west init** outcome:
```
~/D/west-test-project ❯❯❯ west init -m https://gitlab.com/gon1332/west-test-project                             master
Initializing in /home/gon1332/Development/west-test-project
=== Cloning manifest repository from https://gitlab.com/gon1332/west-test-project, rev. master into /home/gon1332/Development/west-test-project/.west/tmp
Initialized empty Git repository in /home/gon1332/Development/west-test-project/.west/tmp/.git/
Username for 'https://gitlab.com': gon1332
Password for 'https://gon1332@gitlab.com':
warning: redirecting to https://gitlab.com/gon1332/west-test-project.git/
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 3 (delta 0), reused 0 (delta 0)
Unpacking objects: 100% (3/3), done.
From https://gitlab.com/gon1332/west-test-project
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
975ce7decd514ef0583814a4d1eff26e861c33f1 refs/remotes/origin/master
Branch 'master' set up to track remote branch 'master' from 'origin'.
Already on 'master'
cloning https://github.com/zephyrproject-rtos/west at revision master
=== Cloning west repository from https://github.com/zephyrproject-rtos/west, rev. master into /home/gon1332/Development/west-test-project/.west/west
Initialized empty Git repository in /home/gon1332/Development/west-test-project/.west/west/.git/
remote: Enumerating objects: 132, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (79/79), done.
remote: Total 2246 (delta 64), reused 71 (delta 37), pack-reused 2114
Receiving objects: 100% (2246/2246), 636.78 KiB | 1.21 MiB/s, done.
Resolving deltas: 100% (1273/1273), done.
From https://github.com/zephyrproject-rtos/west
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
 * [new tag]         v0.1.0     -> v0.1.0
 * [new tag]         v0.2.0     -> v0.2.0
 * [new tag]         v0.2.0rc1  -> v0.2.0rc1
 * [new tag]         v0.2.0rc2  -> v0.2.0rc2
 * [new tag]         v0.2.0rc3  -> v0.2.0rc3
 * [new tag]         v0.3.0     -> v0.3.0
 * [new tag]         v0.3.0rc1  -> v0.3.0rc1
 * [new tag]         v0.4.0     -> v0.4.0
 * [new tag]         v0.4.0rc1  -> v0.4.0rc1
 * [new tag]         v0.5.0     -> v0.5.0
 * [new tag]         v0.5.1     -> v0.5.1
 * [new tag]         v0.5.2     -> v0.5.2
 * [new tag]         v0.5.3     -> v0.5.3
 * [new tag]         v0.5.4     -> v0.5.4
 * [new tag]         v0.5.5     -> v0.5.5
 * [new tag]         v0.5.6     -> v0.5.6
 * [new tag]         v0.5.7     -> v0.5.7
 * [new tag]         v0.5.8     -> v0.5.8
c724ee6a62b6b64ced0e832d0e3198c828406c4f refs/remotes/origin/master
Branch 'master' set up to track remote branch 'master' from 'origin'.
Already on 'master'
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'post-init' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')
```

carlescufi, 2019-05-27:
@gon1332 stick to v0.5.8 of west for now, master is undergoing changes. In your manifest:

```
west:
  url: https://github.com/zephyrproject-rtos/west
  revision: v0.5.8
```

zafersn, 2023-10-07:
Hi @carlescufi 

> @Laczen
>
> Step 1. You create a `west.yml` in your main "downstream" repository, let's assume it's called "foo":
>
> `foo/west.yml`:
>
> ```
> west:
>   url: https://github.com/zephyrproject-rtos/west [...]

thanks for this great tip, it worked for me. I have a related question on that.

According to the getting started guide ` west zephyr-export` command is required to export. So, I wanted to run that but when I type west command there is no zephyr-export cmd. I did some research on having a west command for my project and I developed my own script, so now I can run my command by` west my-own-cmd`. So basically my question is
1. to run zephyr scripts as the west commands now in my case what should I do, I mean how can I run `west zephyr-export`.  is it enough to address the zephyr script in my west-commands.yml file like this?

```
west-commands:
  - file: scripts/west_commands/commands.py
    commands:
      - name: netfeasa-cmd
        class: netCmd
        help: run commands for netfeasa
  - file: C:/zephyrproject/zephyr/scripts/west_commands/export.py
    commands:
      - name: zephyr-export
        class: ZephyrExport
        help: export Zephyr installation as a CMake config package
```
2. And also another question is that if I follow this instruction Most of the Zephyr modules are not installed unless I manually add these modules into my own my-project/west.yml file. I dont know why. Any suggestions will be appreciated. Cheers!

carlescufi, 2023-10-21:
@zafersn the zephyr-export, build and flash commands are part of the zephyr tree, so you need to import the Zephyr tree from your own manifest. About modules not being installed, are you importing the zephyr manifest?

zafersn, 2023-10-23:
@carlescufi thanks for your reply. 

> so you need to import the Zephyr tree from your own manifest.

what do you mean exactly here Can you please elaborate on it?
 you mean that just add the Zephyr into project lists? like this:

```
projects:
.
.
.
.
    - name: nrf
      repo-path: sdk-nrf
      path: nrf
      remote: ncs
      revision: v2.4.2
      import: true
    - name: zephyr
      repo-path: zephyr
      revision: main
      remote: zephyrproject.
      import: true
.
.
.

```

or you mean should I copy all zephyr manifest contents into my manifest file(west.yml)

BTW  I fixed the issue by adding `import: true` keyword into the manifest file. I am asking questions because  I am really wondering if I fixed the issue by doing it correctly or was lucky and everything worked luckily. 

Thank you.

carlescufi, 2023-10-24:
@zafersn you should not import both NCS and Zephyr in your manifest. In my opinion you should only import one or the other, since both come with a zephyr tree.