zephyrproject-rtos/zephyr #15145: HF clock stop fail

pdunaj, 2019-04-03:
**Describe the bug**
Replicated with PCA20041 (probably will be replicated with anything else with nrf52840). Plug and unplug USB. Observe that sometimes we get `HF clock stop fail` with some error =16 or -69. 

**To Reproduce**
Plug/unplug the USB.

**Expected behavior**
HF clock must be stopped.

**Impact**
Showstopper as HF clock consumes power. Current consumption will increase from few uA to few hundreds uA.

**Screenshots or console output**
```
DUT USB OFF
WAITING FOR state:DISCONNECTED.
[00882599] <inf> ble_state: Disconnected from ce:2e:90:f8:63:ef (random) (reason 19)
[00882606] <inf> event_manager: e: ble_peer_event id=0x200025fc DISCONNECTED
[00882608] <inf> ble_adv: Advertising stopped
[00882608] <inf> ble_adv: Use fast advertising
[00882633] <inf> ble_adv: Advertising started
[00882633] <inf> hid_state: Subscriber 0x200025fc disconnected
[00882640] <inf> event_manager: e: led_event led_id:1 mode:BLINKING period:1000 color: < 150 0 0 >
[00890356] <err> usb_nrfx: HF clock stop fail: -16
[00890357] <inf> usb_state: USB HID disabled
[00890360] <inf> event_manager: e: hid_report_subscription_event MOUSE report notification disabled by 0x2000421e
[00890361] <inf> hid_state: Clear report data (0)
[00890363] <inf> event_manager: e: usb_state_event id:0x2000421e state:DISCONNECTED
```

**Environment (please complete the following information):**
 ncs zephyr: `0bf5263b0522bb5cfac84eefdfdee86dc2c67e3e` (upstream `d3bb3cf`)

**Additional context**
N/A

carlescufi, 2019-04-03:
@pdunaj can it be that you are advertising when you unplug, so that the HF clock needs to be kept running for BLE?

pdunaj, 2019-04-03:
Hi @carlescufi , this is possible. I will have a look on my desk. From the logs it seems however that error can come when we are connected.

```
[00432499] <inf> ble_state: Connected to e1:67:21:b0:28:1e (random)
[00432499] <inf> ble_state: Identity 0 has 1 bonds
[00432500] <inf> ble_state: Set security level
[00432506] <inf> event_manager: e: ble_peer_event id=0x200025ec CONNECTED
[00432508] <inf> ble_adv: Advertising stopped
[00432508] <inf> hid_state: Subscriber 0x200025ec connected
[00432513] <inf> event_manager: e: led_event led_id:1 mode:ON period:0 color: < 150 0 0 >
[00432551] <wrn> bt_ctlr_hci: handle: 0x0000, CSA: 0.
[00464739] <inf> event_manager: e: battery_state_event battery IDLE
[00464743] <inf> event_manager: e: led_event led_id:0 mode:ON period:0 color: < 255 0 20 >
[00464963] <err> usb_nrfx: HF clock stop fail: -16
```

carlescufi, 2019-04-03:
@pdunaj well when you are connected the HF clock is also required by the Link Layer often. 
@cvinayak can you weigh in as well?

pawelzadrozniak, 2019-04-03:
Error code -16 means -EBUSY. Looks like something else has requested HFCLK.
It looks like not-really-an-error - there is a primitive reference counter in current implementation of clock driver (nrf_power_clock.c). When a module requests HFCLK to run, that counter is increased and decreased on stop request. -EBUSY is returned when reference counter > 0 which means that someone else is using HFCLK.

Current driver does not track who is requesting the clock, so such behavior would be an error in case if one module requests more than once and stops only once (and symetrically, -EALREADY if requested once and stopped multiple times).
I have spotted the bug in USB shim - HFCLK is stopped on cable detachment and in usb_dc_detach(), so it can potentially disable HFCLK when sth else is using it, but it looks like it's a different case.

pdunaj, 2019-04-03:
I checked the reference counter and who calls clock on/off. It seems that mayfly is constantly enabling and disabling the clock. I expect this is correct and clock is enabled only during the transmission (I have not seen any changes in this area lately). Counter is going up and down periodically so there seems to be no missed stop.

I can't check it now but I will measure power consumption tomorrow. From our point of view this is the important thing, i.e. to stay within power budget while maintaining the connection. Will give the update tomorrow.

As for the log itself, it must be converted from LOG_ERR to something else (inf probably if this is expected behavior) as our CI reacts to all errors. (@pawelzadrozniak )

I also had a look into `nrf_power_clock.c` and there are two things that are potential bugs (@cvinayak ):
- There is no barrier before `m16src_grd` is set to zero (this is commented as release of the guard). As there is no barrier (even compiler barrier) this can be moved around and set to zero before expected activities are done.
- When guard is active start/stop respond with `-EAGAIN`. This assumes caller could retry but it will not always be possible (e.g. when low priority task 'holds' the guard and becomes preempted by caller of start/stop). Mayfly for example is not checking return value at all! It is so potentially dangerous and can lead to leaving the clock in an undesired state.

cvinayak, 2019-04-05:
Historically, the -EALREADY return value in the clock driver _m16src_stop was a fatal ASSERT if there was a miss match of start and stop. Current day, its a return value. You may check this return in controller code and assert. If you do hit this assert, then there is a bug in controller. There is not harm in adding such a check in controller.

Let me think aloud here,

Case 1, where HF clock is not running:
1. m16src_ref is 0
2. _m16src_start is called by USB subsystem
3. Case 1a, while inside _m16src_start with _m16src_grd set (implies _m16src_ref = 1), controller calls _m16src_start
3a. controller will get -EALREADY and _m16src_ref has been increased (_m16src_ref = 2)
3b. As the _m16src_ref is always pre-incremented, there shall not be a case with a request from controller to start HFCLK to fail, hence no check on the return code. Action: You may add a variable to get the return value and assert on -EBUSY in the mayfly_xtal_start to catch any bug in controller.
4. Case 1b, while inside _m16src_start with _m16src_grd set (_m16src_ref = 2), controller calls _m16src_stop in mayfly_xtal_stop and isr_radio_state_close
4a. controller shall only call _m16src_stop when it knows it has called _m16src_start previously in the radio event prepare. Action: You may add a variable to get the return value and assert on -EALREADY in the mayfly_xtal_stop and ist_radio_state_close to catch any bug in controller.

Case 2, When HF clock is running:
1. TODO, at next opportunity I will describe.

pawelzadrozniak, 2019-04-08:
https://github.com/zephyrproject-rtos/zephyr/pull/15257

pdunaj, 2019-04-09:
I have created https://github.com/zephyrproject-rtos/zephyr/issues/15286 for observations regarding the HF clock start/stop issues (guard and lack of barrier).

PR #15257 (2019-04-15): drivers: usb: nordic: Do not log an error if HFCLK is busy
HFCLK may be requested for multiple modules, i.e. Bluetooth stack.
When any other module has requested HFCLK to run, the driver will
return -EBUSY on free attempt which is not an error - thie means
that free request has been processed, but someone else is still
requiring the clock to run. When all clock users free the clock,
is is then stopped to save power. Driver may also return -EAGAIN
when any other module is accesing the driver at the same time.
A proper reaction to -EAGAIN has been implemented here (wait and
try again).

Fixes: # [... cut]
closed 2019-04-15 by carlescufi