zephyrproject-rtos/zephyr #15159: undefined reference to `bsearch'

StefJar, 2019-04-03:
It seems that the "bsearch" function got removed from the stdlib implementation.

A note on "bsearch":

[Despite the name, neither C nor POSIX standards require this function to be implemented using binary search or make any complexity guarantees.](https://en.cppreference.com/w/c/algorithm/bsearch)

I like to use these function and maybe others as well. Is it possible to provide it?

StefJar, 2019-04-03:
I quickly looked up the linux kernel. It brings an own implementation. Maybe a good point to start from

https://github.com/torvalds/linux/blob/master/lib/bsearch.c

balajikulkarni, 2019-04-11:
@StefJar I am interested in taking this one as my first issue. Let me know if I can start working on changes.

StefJar, 2019-04-12:
thx for taking this issue. There are several functions defined but not necessary implemented in the stdlib.

Think a good approach is to have a look how the Linux guys are handling this problem.

IMHO these functions like bsearch, qsort etc. should stay out of the stdlib to have a general small lib footprint.

balajikulkarni, 2019-04-15:
@StefJar You are right. In case of Linux, this is implemented as part of separate **lib**, instead of having it done as part of **stdlib**. 
I see we do have zephyr/lib/libc/minimal/ which includes these kinds of utilities. Would this be the right place to have bsearch implemented as well? If not please suggest the right one.

StefJar, 2019-04-16:
Think from an end-user perspective it only counts, that the linker finds the symbols defined in the stdlib.h.
Usually the linker does dead code elimination. Means that if you compiled a function, but not using it it gets removed. We can use this mechanism and add an extra lib like bsearch.lib, qsort.lib etc. and let the linker do the dirty removal work for us.
Additional we can add a section at kconfig were the build of these lib(s) get activated. They are activated by default. If someone links to bring down his compile/link time he can deactivate this libs.

balajikulkarni, 2019-04-16:
I have the changes ready and tested locally. I have added couple of new files under libc for now (bsearch.h and bsearch.c).
But I didn't get the last part on adding section to kconfig and getting these lib activated by default. Could you please share some reference or some more details on how I can achieve this?

commit a25dce96 (2019-04-26): libc: minimal: Add bsearch function

This function implements generic binary-search.

Fixes #15159
closed 2019-04-26 by andrewboie