zephyrproject-rtos/zephyr #15163: nsim_*_mpu_stack_guard fails if CONFIG_USERSPACE=n but CONFIG_HW_STACK_PROTECTION=y

andrewboie, 2019-04-03:
For example, with tests/posix/common:

```
Console now belongs to UART, hit CRTL-] to return to simulator.
***** Booting Zephyr OS v1.14.0-rc3-100-g94b8f14eca13 *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
Bounce test OK
Barrier test OK
PASS - test_posix_pthread_execution
===================================================================
starting test - test_posix_pthread_termination
Thread 0 starting with a priority of 4
Cancelling thread 0
Thread 1 starting with a priority of 3
Cancelling thread 1
Thread 1 could not be cancelled
Exception vector: 0x6, cause code: 0x2, parameter 0x4
Address 0x8000bbc0
***** Stack Check Fail! *****
Current thread ID = 0x800003ec
Fatal fault in thread 0x800003ec! Aborting.
Thread 3 starting with a priority of 1
Cancelling thread 3
Thread 3 could not be cancelled

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/posix/common/src/pthread.c:419: test_posix_pthread_termination: (ret not equal to ESRCH)
cancelled a terminated thread!
FAIL - test_posix_pthread_termination
===================================================================
starting test - test_posix_multiple_threads_single_key

Different threads set different values to same key:
thread 0: set value = -2147479816 and retrieved value = -2147479816
thread 1: set value = -2147479560 and retrieved value = -2147479560

PASS - test_posix_multiple_threads_single_key
===================================================================
starting test - test_posix_single_thread_multiple_keys

Single thread associates its value with different keys:
Exception vector: 0x6, cause code: 0x2, parameter 0x4
Address 0x8000d3e4
***** Stack Check Fail! *****
Current thread ID = 0x8000064c
Fatal fault in thread 0x8000064c! Aborting.

```

andrewboie, 2019-04-04:
To reproduce, set CONFIG_TEST_USERSPACE=n in tests/posix/common
Seems to reproduce very reliably on nsim_em_mpu_stack_guard

I spent some time looking at it today, but the root cause is still unclear. The memory exception does not seem to be against the MPU region currently defined for the stack guard, but I need to gather more data.

vonhust, 2019-04-04:
@andrewboie the root cause for nsim_em_mpu_stack_guard is that mpu_stack_guard also requires power of 2 stack allocation.  
In arc/arch.h,  I consider this only for the case of USERPACE.  I will fix it soon.

andrewboie, 2019-04-05:
> also requires power of 2 stack allocation.

@wentongwu It should not. Stack sizes should be arbitrary.
We're only setting an MPU region for the guard area for supervisor mode stacks.
So the stack buffer just needs to be aligned to the size of the guard region.

```
| guard area       |             stack buffer              |
|--2K MPU REGION --|
```

We don't need to set an MPU region for the stack buffer itself, a supervisor mode thread should already be able to read it.

The code appears to correctly align stack objects to 2K.

PR #15248 (2019-04-09): arch: arc: fix the bug of mpu stack guard when no USERSPACE
* fix the stack allocation and initialization
  for mpu stack guard when USERSPACE is not configured
* fixes #15163
commit 290aa59a (2019-04-09): arch: arc: fix the bug of mpu stack guard when no USERSPACE

* fix the stack allocation and initialization
  for mpu stack guard when USERSPACE is not configured
* fixes #15163
closed 2019-04-09 by nashif