zephyrproject-rtos/zephyr #15166: [Coverity CID :197537]Control flow issues in /subsys/power/power.c

aasthagr, 2019-04-03:
Static code scan issues seen in File: /subsys/power/power.c
 Category: Control flow issues
 Function: _sys_suspend
 Component: Other
 CID: 197537
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-04-04:
```
________________________________________________________________________________________________________
*** CID 197537:  Control flow issues  (DEADCODE)
/subsys/power/power.c: 118 in _sys_suspend()
112     		}
113     #endif
114     		/*
115     		 * Disable idle exit notification as it is not needed
116     		 * in deep sleep mode.
117     		 */
>>>     CID 197537:  Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "_sys_pm_idle_exit_notificat...".
118     		_sys_pm_idle_exit_notification_disable();
119     	}
120     
121     	/* Enter power state */
122     	sys_pm_debug_start_timer();
123     	sys_set_power_state(pm_state);

** CID 197536:  Parse warnings  (PARSE_ERROR)
/include/mgmt/buf.h: 11 in ()

```

nashif, 2019-04-04:
lower prio based on the nature of issue.

andyross, 2019-04-04:
Drive-by analysis: this looks like a false positive.  If we don't have any deep sleep states configured, then sys_pm_is_deep_sleep_state() will always return false, thus making the local deep_sleep variable always false, thus preventing this block from executing.

But that's static configuration.  I don't see anything wrong with the code as written (beyond the fact that we have outrageously poor coverage of the power management stuff in general and lots of it does indeed seem to have been written speculatively for platforms that don't quite exist  yet).

Suggest closing.

nashif, 2019-04-09:
false positive

closed 2019-04-09 by nashif