zephyrproject-rtos/zephyr #15167: [Coverity CID :197536]Parse warnings in /include/mgmt/buf.h

aasthagr, 2019-04-03:
Static code scan issues seen in File: /include/mgmt/buf.h
 Category: Parse warnings
 Function: 
 Component: Other
 CID: 197536
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-04-04:
```
** CID 197536:  Parse warnings  (PARSE_ERROR)
/include/mgmt/buf.h: 11 in ()

________________________________________________________________________________________________________
*** CID 197536:  Parse warnings  (PARSE_ERROR)
/include/mgmt/buf.h: 11 in ()
5      */
6     
7     #ifndef ZEPHYR_INCLUDE_MGMT_BUF_H_
8     #define ZEPHYR_INCLUDE_MGMT_BUF_H_
9     
10     #include <inttypes.h>
>>>     CID 197536:  Parse warnings  (PARSE_ERROR)
>>>     cannot open source file "cbor_encoder_writer.h"
11     #include "cbor_encoder_writer.h"
12     #include "cbor_decoder_reader.h"
13     struct net_buf;
14     
15     struct cbor_nb_reader {
16     	struct cbor_decoder_reader r;

```

nashif, 2019-04-04:
invalid setup, looks like west is not setup correctly and does not load modules. @aasthagr, you need to fix this.

closed 2019-04-04 by nashif