zephyrproject-rtos/zephyr #15178: samples/mpu/mem_domain_apis_test:  Did not get to "destroy app0 domain", went into indefinite loop

cinlyooi-intel, 2019-04-04:
**Describe the bug**
According to sample.yaml,  I am suppose to get console output "destroy app0 domain", instead, the test appears to go into indefinite loop

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-104-g4aa48833d812 (delayed boot 1000ms) *****
initialize memory domains
add app thread 0 (0x204001e0) into app0_domain.
add app thread 1 (0x20400264) into app1_domain.
add app thread 2 (0x204002e8) into app0_domain.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) starts.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) starts.
app thread 1 (0x20400264) yields.
app thread 2 (0x204002e8) starts.
app thread 2 (0x204002e8) yields.
main thread (0x20400500) is back.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) is back.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) is back.
`***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-104-g4aa48833d812 (delayed boot 1000ms) *****
initialize memory domains
add app thread 0 (0x204001e0) into app0_domain.
add app thread 1 (0x20400264) into app1_domain.
add app thread 2 (0x204002e8) into app0_domain.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) starts.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) starts.
app thread 1 (0x20400264) yields.
app thread 2 (0x204002e8) starts.
app thread 2 (0x204002e8) yields.
main thread (0x20400500) is back.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) is back.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) is back.
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-104-g4aa48833d812 (delayed boot 1000ms) *****
initialize memory domains
add app thread 0 (0x204001e0) into app0_domain.
add app thread 1 (0x20400264) into app1_domain.
add app thread 2 (0x204002e8) into app0_domain.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) starts.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) starts.
app thread 1 (0x20400264) yields.
app thread 2 (0x204002e8) starts.
app thread 2 (0x204002e8) yields.
main thread (0x20400500) is back.
main thread (0x20400500) yields.
app thread 0 (0x204001e0) is back.
app thread 0 (0x204001e0) yields.
app thread 1 (0x20400264) is back.
```
**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain zephyr
 - Commit SHA or Version used: 4aa48833d812ca56b62226462d48da24fb9bf756

andrewboie, 2019-04-04:
Another very old test. I'll look into it.

ioannisg, 2019-04-04:
It looks, again, like a problem with the watch-dog IMHO

In addition to that, I agree, this test needs re-work. To name a reason for that, the indicated log output does not correspond to the current MPU driver function calls, at least for ARM.

ioannisg, 2019-04-04:
The sample runs in nrf52 boards, without problems.

ioannisg, 2019-04-04:
Looking more into the sample, I realize that the actual logging from the MPU module is not printed in console, because the logging thread does not get chance to execute (all other threads have higher priorities and, simply, yield to each other).

I am submitting a patch to fix this

andrewboie, 2019-04-04:
I think a better approach is to look at what scenarios this test is covering vs. what is being tested with ztest under tests/kernel/mem_protect, and if there is nothing unique, delete it.

PR #15269 (2019-04-09): samples: remove mem_domain_apis_test
We have actual tests now under tests/kernel/mem_protect/userspace
and tests/kernel/mem_protect/mem_protect that exercise all the
memory domain APIs. This old test is superfluous.

Fixes: #15178
commit 9f218870 (2019-04-09): samples: remove mem_domain_apis_test

We have actual tests now under tests/kernel/mem_protect/userspace
and tests/kernel/mem_protect/mem_protect that exercise all the
memory domain APIs. This old test is superfluous.

Fixes: #15178
closed 2019-04-09 by nashif