zephyrproject-rtos/zephyr #15180: testing flash driver clients in CI

nvlsianpu, 2019-04-04:
**problem**
Component which use flash driver are not well tested i CI. Test for these component are on-board tests, so are not run in CI.

components scope:
* subsys/settings
* subsys/FS/fcb
* subsys/FS/nvs
* subsys/dfu
* ...

**solution**
We can provide emulated flash driver https://github.com/zephyrproject-rtos/zephyr/pull/8254 (https://github.com/zephyrproject-rtos/zephyr/issues/7403) which will be used along with qemu's as test target - so it can be very easily to run these tests in zephyr CI.

some component already have test using RAM emulated driver (like NFFS) - this prove that this enchantment is not difficult to achieve.

nvlsianpu, 2019-04-04:
cc @carlescufi @kapi-no

ioannisg, 2019-08-09:
@nvlsianpu has this issue been worked for 2.0 release? Otherwise, can we move this to next one?

nvlsianpu, 2019-08-09:
Both posix flash driver and emulated flash driver for qemu are available in codebase since while.
FCB, NVS, settings and dfu/ modules  are tested in CI using qemu and native_posix targets.

closed 2019-08-09 by nvlsianpu