zephyrproject-rtos/zephyr #15181: ztest issues

eibe85, 2019-04-04:
Scenario is using UART on nrf91 for printing on PC side. All zasserts and the the "regular" printouts of tests like what test is running and if it has passed is skipped or has failed compete for the UART.
Some of the test cases finish fast and some slow. If a test case finishes fast it looks like it can disturb printouts of previous tests.
As far as I can see from the ztest.c file tests are executed in a separate thread. It looks like there are unhanded race conditions between the thread that handles the printout of the tests and the test tread. If a test runs to fast (e.g. is skipped) the printout of the previous tests case is stopped before it has finished. I also think there are unhandled concurrency issues between the test case printouts and the printout of test case asserts.
I suspect this is less likely to occur on a Linux box not printing over slow UART, but with UART it happens quite often.

rljordan, 2019-04-05:
@eibe85 Can you please provide details on how to reproduce this?  Can we get specific changes to the concurrency?

eibe85, 2019-04-05:
@rljordan I am running a test sute wih multiple test cases the UART baud rate is set to 115200. If I use ztest_test_skip() on multiple consecutive test cases to skip those I can observe that only the last skip printout is represented properly. Here is an example output:
"
============= - test_socket_connect_tcp_ipv6
starting test - test_socket_connect_udp_ipv6
SKIP - test_socket_connect_udp_ipv6
=============================================amily:1 type:1 protocol:6
"
Both of the above tests have been set to be skiped using ztest_test_skip() as the first part of the test case function.
I don't think I can upload the test code or the script that handles the UART. I would have to check on that. But I suspect that if any two consecutive tests are set to be skiped they would fail in this way. 

I also tried setting a half a second k_sleep as the first part of the ztest_test_skip() and then the printout is correct:
"
===================================================================
starting test - test_socket_connect_tcp_ipv6
SKIP - test_socket_connect_tcp_ipv6
===================================================================
starting test - test_socket_connect_udp_ipv6
SKIP - test_socket_connect_udp_ipv6
===================================================================
"
But I do not see this as a good solution only a workaround. The tests that execute fast still manages to mangle the print output. I don't think the test cases should handle this I think it should be handled in the ztest.c file by waiting for a printout to finish before starting a new test. Waiting for the printouts to finish should also ensure that any printouts in a test case is not mangled or mangles the test case pass, fail, start msg, etc. by the two threads (test manager thread and test case) stepping on each other.

andrewboie, 2019-04-08:
printk() is synchronous.

whats printing "amily:1 type:1 protocol:6"?

andrewboie, 2019-04-09:
>  and it looks like it uses printf

It is not, the PRINT macro maps to printk():

```
ztest/include/ztest.h
44-/* FIXME: Properly integrate with Zephyr's arch specific code */
45-#define CONFIG_X86 1
46:#define CONFIG_PRINTK 1
47-#endif
48-
49-#include <misc/printk.h>
50:#define PRINT printk
51-
52-#include <zephyr.h>
```

There's also a PRINT_DATA() macro in ztest which maps to printk:

```
include/tc_util.h
16-#include <misc/printk.h>
17-
18:#define PRINT_DATA(fmt, ...) printk(fmt, ##__VA_ARGS__)
19-
20-#if defined CONFIG_ARCH_POSIX
```

and TC_PRINT() which is mapped onto PRINT_DATA():

```
75:#define TC_PRINT(fmt, ...) PRINT_DATA(fmt, ##__VA_ARGS__)
```

I don't know why PRINT_DATA() doesn't just map onto PRINT.

Would this solution be acceptable:

1) #define PRINT PRINT_DATA
2) Implement PRINT_DATA as an inline function which grabs a global sys_mutex before it prints something.

This will not prevent console clobbering from test cases that use printk() directly, but I think serializing printk() itself with a sys_mutex might be too big of a change. The tests could be changed to use TC_PRINT() instead of printk() though.

eibe85, 2019-04-11:
You are right about the print macro. I had just looked at the PRINT_LINE macro but at one defined in the wrong place where it was directed to use printf. The one I looked at is not used in ztest.

I think your solution would improve things considerably. It should a low all lines to be printed completely then. At least if the output is parsed you should then get the correct number of SKIP, FAIL and PASS

eibe85, 2019-05-20:
I think perhaps I was wrong. I was looking at implementing the fix as andrewboie described but that turned out harder than expected to get trough the CI. However wen digging into the code I think I found that the problem is in the use of DMA on the UART and the printk function. It looks like the print buffer could be modified while the DMA is running and that there is no error handling if the program tries to write while a DMA -> UART write operation is in progress. I will try to get this confirmed and close the issue.

eibe85, 2019-05-20:
Unfortunately I again looked at the wrong code so the bug still stands.

nashif, 2020-03-12:
stale issue

closed 2020-03-12 by nashif

WayneKeenan, 2022-10-21:
Hi, 

Using TC_PRINT in unit tests is still causing a clobbering issue. The assert messages, per test results and the summary can all be truncated.

Env:
Zephyr: v2.7.99-ncs1
nRF Connect v1.9.0
HW: nrf52833 HW
DBG:SWD / RTT 

However, I found a workaround is to add this to `prj.conf`

```
CONFIG_USERSPACE=y
```