zephyrproject-rtos/zephyr #15193: tests/net/socket/getaddrinfo: assert logic is backwards

mike-scott, 2019-04-04:
**Describe the bug**
Assert logic in https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/net/socket/getaddrinfo/src/main.c#L23 is backwards.  You can run this test without access to a local dnsmasq server and it passes.

**To Reproduce**
1. Make sure QEMU net-tools are started, **but no local dnsmasq server is available**.
2. cd tests/net/socket/getaddrinfo
3. mkdir build && cd build
4. cmake -DBOARD=qemu\_x86 ..
5. make run
6. test passes(?!?)

**Expected behavior**
Test should fail if no local dnsmasq server is running

**Impact**
Test is not working

**Screenshots or console output**
```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc3-297-g6fcf986e7775 *****
Running test suite socket_getaddrinfo
===================================================================
starting test - test_getaddrinfo_ok
PASS - test_getaddrinfo_ok
===================================================================
starting test - test_getaddrinfo_no_host
PASS - test_getaddrinfo_no_host
===================================================================
Test suite socket_getaddrinfo succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.10.0
 - SHA: https://github.com/zephyrproject-rtos/zephyr/commit/0734b4fcaaea62e5f887d932a79b31507d61429d

mike-scott, 2019-04-04:
I have a feeling more of the socket tests have this same issue.

jukkar, 2019-04-04:
Please note that all the programs under `tests/net` are suppose to be self contained, they are not suppose to connect to outside world. The programs in `samples/net/` are different in this respect as those ones will need outside connectivity to function properly.

That said, I tried to run this test for native_posix and it fails with this message
`ASSERTION FAIL [net_if_get_link_addr(iface)->addr != ((void *)0)] @ zephyr/subsys/net/ip/net_if.c:2870`
I will investigate what is going on here as the test is suppose to be run in native_posix according to its testcase.yaml file. I will also look what the test is actually doing or trying to do.

mike-scott, 2019-04-04:
@jukkar can you try and run this test for qemu_x86?  Without any setup of local dnsmasq it will PASS.  Which is not correct.  Take a look here at the assert checks:
https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/net/socket/getaddrinfo/src/main.c#L23

mike-scott, 2019-04-04:
BTW, if you look at the asserts, there is clearly a bug there.  Removing the `bug` tag doesn't change this.  Are we to ship LTS with incorrect tests?

mike-scott, 2019-04-04:
Ugh.  Re-reading this.  I guess this test is written as a failure?  Why does this even exist?  Samples would do compile testing for sanity.

This test is a waste of CPU cycles.

I'll close, but this is silly.

mike-scott, 2019-04-04:
To clarify why this test is silly:
The whole reason I started investigating this, is that an actual DNS query sent via the Socket-based APIs which is never returned by the server will hang the Zephyr device.  We have an edge gateway setup where containers are loaded which handle NAT64, DNS64, joining BLE 6lowpan devices as well as OpenThread.  This issue was identified when the DNS64 container spins up a bit later than the BLE joiner.   The perfect example of this would be when the gateway software is updated and it requires a restart, but the IoT nodes are up and running at the time.

The K_FOREVER semaphore taken here and a bit lower are the cause of the hang: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/sockets/getaddrinfo.c#L119

This test has no real network setup, so DNS subsys lib returns EAI_CANCELLED when the query cannot be sent.  This test checks exact that: that a query cannot be sent and nothing else.

jukkar, 2019-04-05:
Lets not close this as there is an issue. As I said, I will investigate how to fix it.

jukkar, 2019-04-05:
There are too may assignees so setting only one. @mike-scott, please set only one assignee at a time as otherwise it is not clear who will start to fix the issue.

mike-scott, 2019-04-05:
@jukka Apologies.  I'm too used to PRs where that spot is for reviews.  Thank you for taking a look here and at the new bug I opened.

pfalcon, 2019-04-16:
> This test has no real network setup, so DNS subsys lib returns EAI_CANCELLED when the query cannot be sent. This test checks exact that: that a query cannot be sent and nothing else.

Well, it tests at least something, specifically what it could (easily) test. And for what it tested, the logic is correct, so updating the ticket title.

PR #15208 (2019-04-30): Elaborate getaddrinfo() test, add mock server to actually receive and verify DNS query sent by getaddrinfo() call
Fixes #15193
commit 960e6d95 (2019-04-30): tests: net: getaddrinfo: Fix test to check the query is sent

As the getaddrinfo() test is suppose to be self contained, then
we will need to install handler for the sent DNS query. After we
receive the query in process_dns() thread, we verify that the
query was sent properly. We do not return DNS reply back to
the caller in this case as we do not want to create DNS server
just for this simple test application. This can be changed later
if we get DNS server functionality in Zephyr network stack.

Fixes #15193
closed 2019-04-30 by andrewboie