zephyrproject-rtos/zephyr #15196: logging: Support for blocking deferred logging

jimparis, 2019-04-04:
With deferred logging, messages get dropped when buffers are full.  In many cases, I would prefer if the `LOG_*` caller would block instead, when possible (thread context).  For example, something like this, maybe hidden behind a config option `CONFIG_LOG_BLOCKING`:

```diff
--- a/include/logging/log_msg.h
+++ b/include/logging/log_msg.h
@@ -319,7 +319,8 @@ union log_msg_chunk *log_msg_no_space_handle(void);
 static inline union log_msg_chunk *log_msg_chunk_alloc(void)
 {
        union log_msg_chunk *msg = NULL;
-       int err = k_mem_slab_alloc(&log_msg_pool, (void **)&msg, K_NO_WAIT);
+       int err = k_mem_slab_alloc(&log_msg_pool, (void **)&msg,
+                                   k_is_in_isr() ? K_NO_WAIT : K_FOREVER);
 
        if (err != 0) {
                msg = log_msg_no_space_handle();
```

Note that this isn't about immediate vs deferred logging -- I still want logs to be deferred when possible, I just want to ensure that the message gets queued instead of dropped.

aescolar, 2019-04-08:
@jarz-nordic @nordic-krch

nordic-krch, 2019-05-16:
@jimparis sounds reasonable. I am afraid about the case when log is called from the thread which is part of log processing path, e.g logs in shell instance. Then `K_FOREVER` would block logging and user is left in blind, hard to for him to determine the reason. One option would be to no use `K_FOREVER` to something limited (e.g. 100ms). User will experience logger choking and that could be a hint which is better than nothing.

PR #16194 (2019-09-24): logging: Add option to block in thread context
Added CONIFG_LOG_BLOCK_IN_THREAD option to block until buffer for
log message is available. When log message is called in the thread
and there is no buffer available in the pool, thread will block with
configurable timeout (CONFIG_LOG_BLOCK_IN_THREAD_TIMEOUT_MS). If
buffer cannot be allocated by that time, message will be dropped.

Fixes #15196.
closed 2019-09-24 by andrewboie