zephyrproject-rtos/zephyr #15197: Socket-based DNS API will hang device if DNS query is not answered

mike-scott, 2019-04-04:
**Describe the bug**
Using the socket-based DNS query APIs located here: 
If a query is generated and sent, but never returned, the Zephyr device will stay in that state forever.

The cause of this is the K_FOREVER here and a bit lower (in the IPV6 code): The whole reason I started investigating this, is that an actual DNS query sent via the Socket-based APIs which is never returned by the server will hang the Zephyr device. We have an edge gateway setup where containers are loaded which handle NAT64, DNS64, joining BLE 6lowpan devices as well as OpenThread. This issue was identified when the DNS64 container spins up a bit later than the BLE joiner. The perfect example of this would be when the gateway software is updated and it requires a restart, but the IoT nodes are up and running at the time.

The K_FOREVER semaphore taken here and a bit lower are the cause of the hang: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/sockets/getaddrinfo.c#L119

**To Reproduce**
This is harder to reproduce with native Zephyr samples, because most of them do not use the Socket-based APIs for DNS, and if they do use them, they are natively setup for running in QEMU which has no routing behavior and requires modification for actual networking hardware like FRDM-K64F, etc.

**Expected behavior**
Device should never hang forever.  Pick an expected timeout time and let's generate a timeout so the caller of the DNS function can decide how to react.

**Impact**
Any device using the socket-based APIs for DNS can be easily hung if the UDP packet for the DNS query is lost, etc.  Also if DNS entry is not correct, developer will never get any notification other than an unresponsive device.

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.10.0
 - Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/0734b4fcaaea62e5f887d932a79b31507d61429d

mike-scott, 2019-04-04:
@pfalcon @jukkar Testing ignored.  This is the bug that I would like to see fixed.

If you would like pointers on how to test, please let me know and I'll setup instructions for a native sample of your choice using whatever real hardware with networking you happen to have.

rljordan, 2019-04-05:
@jukkar  Can you please assess and add a priority to this?

jukkar, 2019-04-05:
Setting only one assignee.

PR #15210 (2019-04-11): net: sockets: Make sure that getaddrinfo() cannot hang forever
If for some reason the DNS resolver callback is not called properly
then make sure that semaphore will not block forever.

Fixes #15197
commit 1897896e (2019-04-11): net: sockets: Make sure that getaddrinfo() cannot hang forever

If for some reason the DNS resolver callback is not called properly
then make sure that semaphore will not block forever.

Fixes #15197
closed 2019-04-11 by nashif