zephyrproject-rtos/zephyr #15205: hci_usb not working on v1.14.0rc3 with SDK 0.10.0

Matt90o, 2019-04-05:
**Describe the bug**
The HCI_usb example in the zephyr repo stopped working on v1.14.0rc3 using SDK0.10.0. The USB dongle does not show as a USB device. Previous build based on v1.14.0rc2 and SDK 0.9.0 works succesfully. I am using the nrf52840_pca10059 board (dongle).

**To Reproduce**
Steps to reproduce the behavior:
1. Build the HCI_usb example for nrf52840_pca10059 board
2. Flash the hex on the USB dongle
2. Plug in the USB dongle into machine
3. Observe dmesg output: Board does not show up as USB device

**Expected behavior**
After building and flashing the device it should show up as ZEPHYR device in dmesg.

**Impact**
This results in the example not being able to run at all. 

**Screenshots or console output**
-

**Environment (please complete the following information):**
 - OS: Linux in VM (4.19.30-1-MANJARO #1 SMP PREEMPT Tue Mar 19 17:49:12 UTC 2019 x86_64 GNU/Linux)
 - Zephyr v1.14.0rc3 with Toolchain Zephyr SDK 0.10.0
 - Tagged rc3 version used and official SDK 0.10.0

**Additional context**
Please check the general slack here: https://zephyrproject.slack.com/archives/C18PLHN8H/p1554456501803600 . Seems to be a general problem as multiple people have reported seeing this issue.

carlescufi, 2019-04-05:
@pawelzadrozniak @lemrey @barsok is this an issue only on pca10059,  or also on pca10056? I wonder if it's related to the board or the USB stack in fact

jfischer-phytec-iot, 2019-04-05:
sample works fine on reel board and nrf52840_pca10056 (latest master), but not the combination of mcuboot and hci_usb

[00:00:00.201,202] <inf> mcuboot: Jumping to the first image slot
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x96f3b83d
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x733a

addr2line -e build/nrf52840_pca10056/mcuboot/zephyr/zephyr.elf 0x733a
/home/john/zephyr-project/zephyr/drivers/flash/soc_flash_nrf.c:554

carlescufi, 2019-04-05:
@nvlsianpu can you take a look at the above log?

jfischer-phytec-iot, 2019-04-05:
@carlescufi I do not think it is a flash issue, now I got `<err> kernel: Never should get here`

nvlsianpu, 2019-04-05:
codec Line pointed is the statement for write data to flash - so looks like stack corruption.

carlescufi, 2019-04-08:
@Matt90o You are not using MCUboot here but the Nordic stock bootloader right?

Matt90o, 2019-04-08:
> 
> 
> @Matt90o You are not using MCUboot here but the Nordic stock bootloader right?

Yes I am using the Nordic stock bootloader. I am packaging the HCI usb example exactly as described in the docs [here](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10059/doc/index.html#flashing-with-a-device-firmware-update-dfu).

carlescufi, 2019-04-08:
@Matt90o I cannot reproduce this at all with the current master (7cdfe7f26f4bc840a0c3554732b8cf6ed1a645c8) and SDK 0.10.0. The app boots correct and is detected by BlueZ on the Linux Host.

Matt90o, 2019-04-08:
@carlescufi This seems to be fixed in https://github.com/zephyrproject-rtos/zephyr/commit/e01662d387c0dfb5e68ed964174106e57c778fc5. I have tried the HCI usb example again with SDK 0.10.0 and the device pops up again as usb device. So this issue can be closed.

closed 2019-04-08 by Matt90o