zephyrproject-rtos/zephyr #15213: cmake infrastructure in code missing file level license identifiers

kestewart, 2019-04-05:
**Describe the bug**
Using FOSSology to do a test clearing the zephyr-master repo,  its clear we've got a systemic failure to add license information in the CMakeLists.txt files. 

**To Reproduce**
See the following files:
zephyr-master/arch/CMakeLists.txt 
zephyr-master/arch/arc/CMakeLists.txt
zephyr-master/arch/arc/core/CMakeLists.txt
zephyr-master/arch/arc/core/mpu/CMakeLists.txt

zephyr-master/arch/arm/CMakeLists.txt
zephyr-master/arch/arm/core/CMakeLists.txt
zephyr-master/arch/arm/core/cortex_m/CMakeLists.txt
zephyr-master/arch/arm/core/cortex_m/cmse/CMakeLists.txt
zephyr-master/arch/arm/core/cortex_m/mpu/CMakeLists.txt
zephyr-master/arch/arm/core/cortex_m/tz/CMakeLists.txt

zephyr-master/arch/common/CMakeLists.txt

zephyr-master/arch/nios2/CMakeLists.txt
zephyr-master/arch/nios2/core/CMakeLists.txt

zephyr-master/arch/posix/CMakeLists.txt

zephyr-master/arch/riscv32/CMakeLists.txt
zephyr-master/arch/riscv32/core/CMakeLists.txt

zephyr-master/arch/x86/CMakeLists.txt
zephyr-master/arch/x86/core/CMakeLists.txt

zephyr-master/arch/x86_64/CMakeLists.txt
zephyr-master/arch/x86_64/core/CMakeLists.txt

zephyr-master/arch/xtensa/core/CMakeLists.txt
zephyr-master/arch/xtensa/core/startup/CMakeLists.txt

zephyr-master/boards/arc/em_starterkit/CMakeLists.txt

... have stopped here,  as its pretty clear its systemic and needs to be fixed at that level. 

**Expected behavior**
Expect to see comment with 
SPDX-License-Identifier: Apache-2.0
in each of the files, like the rest of the repository.

**Impact**
Showstopper.   Licensing needs to be clean and comprehensive before LTS is released, as this will impact all projects trying to generate an accurate SBOM,  as well as impact CII Gold badge status. 

**Environment (please complete the following information):**
 - Version used:  1.14-rc3

kestewart, 2019-04-05:
Also have found: zephyr-master/cmake/backports/FindPythonInterp.cmake has BSD-3-Clause reference.   

# Distributed under the OSI-approved BSD 3-Clause License.  See accompanying
# file Copyright.txt or https://cmake.org/licensing for details.

So there is some licensing bits.

zephyr-master/cmake   should be included in the cleanup.

SebastianBoe, 2019-04-05:
Do you need to license infrastructure code? I thought it was only the sources that constitute the resulting firmware that needed to be licensed, and not the scripts that control how the firmware is built.

Or is this about over-licensing so that it is easier to for the tool to prove that all files are correctly licensed?

nashif, 2019-04-05:
the above example is taken from somewhere else which requires the license header. For something like CMakeFile.txt files we should be covered by the top level license file. 

From the apache 2.0 license:

```
The 2.0 version of the Apache License was approved by the ASF in 2004. The goals of this license revision have been to reduce the number of frequently asked questions, to allow the license to be reusable without modification by any project (including non-ASF projects), to allow the license to be included by reference instead of listed in every file, 
```

see `"to allow the license to be included by reference instead of listed in every file"`

kestewart, 2019-04-05:
If we have license headers in all the files, we're clear and its a lot easier to figure out automatically and not require manual judgement.    Challenge is some files are imported from other projects,  and the license of those projects needs to be traced back and explicitly documented, unless its in the file.   (see cmake/backports/FindPythonInterp.cmake - and there are others in that directory, which were likely copied from other places).   Its easiest in the long run if its an explicit decision. 

From: https://en.wikipedia.org/wiki/Source_code
Moderately complex software customarily requires the compilation or assembly of several, sometimes dozens or maybe even hundreds, of different source code files.   In these cases, instructions for compilations, **such as a Makefile, are included with the source code**. These describe the programming relationships among the source code files, and contain information about how they are to be compiled.   

CMakeFile.txts seem to fit the above definition,   and leaving them out will cause problems with the CII gold badge status where: 
The project MUST include a license statement in each source file. This MAY be done by including the following inside a comment near the beginning of each file: SPDX-License-Identifier: [SPDX license expression for project]. [license_per_file] 
and we've documented.  "All files contain a SPDX license statement."

nashif, 2019-04-06:
@kestewart:

I am personally fully ok with adding SPDX everywhere and be explicit instead of implicit. Not worth getting into debates about the definition of what source code is and quoting Wikipedia for that :-)

Let's get everyone onboard next week and make this happen.

kestewart, 2019-04-06:
Thanks Anas.      It will make it much easier for everyone doing licensing clearing for products to not have to look into all the cases one by one when license aren't present.

PR #15238 (2019-04-07): License cleanup
Fixes #15235
Fixes #15213
Fixes #15234
closed 2019-04-07 by nashif