zephyrproject-rtos/zephyr #15214: Enforce correct compilers in boilerplate.cmake

DC11011100, 2019-04-05:
**Is your enhancement proposal related to a problem? Please describe.**
I propose enhancing toolchain checkers in boilerplate.cmake. On my system, the calling order project(....) and include(....boilerplate.cmake) in CMakeLists.txt determines whether my build will fail of succeed. CMake's documentation on 'project()' isn't clear as to why this build difference occurs, but its noticeable in the cmake/make outputs. See the attached good/bad files which contain outputs along the process of building. The single
[bad.CMakeCache.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048592/bad.CMakeCache.txt)
[bad.CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048593/bad.CMakeLists.txt)

[good.CMakeCache.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048594/good.CMakeCache.txt)
[good.CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048595/good.CMakeLists.txt)

 change that causes it is found in the attached CMakeLists.txt where I swap the order project() and boilerplate include calls. You'll notice an immediate difference just in the (good/bad).cmake.out files wherein some extra cmake checkers occur in the bad case. You'll also notice the difference in CMakeCache.txt where, for example, the good output CMAKE_CXX_COMPILER_AR and other cmake vars correctly points to my arm embedded toolchain, and in the bad outputs it points to my system's default compilers.

I encountered this problem when writing my own CMakeLists.txt for my own project which uses zephyr kernel, and it wasn't obvious that the cause was call order of project() and include boilerplate.cmake in my CMakeLists.txt

**Describe the solution you'd like**
Reinforce usage of user expected toolchain in boilerplate.cmake by updating the diffed cmake vars. See attached (good/bad).CMakeCache.txt. That way, if they do include boilerplate after project(), the vars are updated before make.

**Describe alternatives you've considered**
Make a clear comment in the sample CMakeLists.txt so this nuance is obvious

**Additional context**
* My system: Linux cocoa 5.0.4-200.fc29.x86_64 #1 SMP Mon Mar 25 02:27:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
* Sample: shell_module
* Board: nrf52840 w/ cpu: pca10056
* Zephyr version: 1.13
* zephyr-env sourced
* clean build dir each time

[bad.CMakeCache.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048603/bad.CMakeCache.txt)
[bad.CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048604/bad.CMakeLists.txt)
[good.CMakeCache.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048605/good.CMakeCache.txt)
[good.CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048606/good.CMakeLists.txt)
[bad.cmake-out.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048611/bad.cmake-out.txt)
[bad.make-out.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048612/bad.make-out.txt)
[good.cmake-out.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048613/good.cmake-out.txt)
[good.make-out.txt](https://github.com/zephyrproject-rtos/zephyr/files/3048614/good.make-out.txt)

aescolar, 2019-04-08:
CC @SebastianBoe

SebastianBoe, 2019-04-08:
placing project() before include() is not supported.

If there is a clean way to detect and report this user-error then a patch to this effect would be very welcome. I would prefer to not add support for placing 'project' before 'include' unless there is a good reason to support it.

gmarull, 2022-08-11:
boilerplate.cmake is no longer used, closing.

closed 2022-08-11 by gmarull