zephyrproject-rtos/zephyr #15223: optimize MPU gap-filling

andrewboie, 2019-04-05:
This is applicable to ARC MPUv3 and ARMv8, whose MPUs do not allow overlapping regions.

Both MPUs implement logic to fill in gaps between user-writable regions with kernel-writable regions, as the "background" MPU setting does not provide access at all.

However, all the gap-filling computation is taking place at context switch time. Even worse, on ARC, the gap-filling mechanism is constantly reprogramming MPU regions for every source region programmed (MPU stack guard, user thread stack, every memory domain partition), and these are MMIO registers, not cacheable memory. This all takes place with interrupts locked, affecting the latency of the system.

We have got to get this logic out of context switching, it does not need to be there. We could add an array of per-thread MPU region settings for every volatile MPU region (i.e. regions change after the system is started.)  to thread.arch. These settings would get re-computed when memory domain parameters change, rather than every context switch. The context switch logic can then be dead simple: just reprogram the volatile MPU regions on context switch with the contents of the array in thread.arch.

If at all possible, it would be great if we could make the gap-filling computation common code, and not implemented for every arch that needs it.

andrewboie, 2019-04-09:
somewhat related task for MMU-based x86: #13003

wentongwu, 2019-04-11:
@ioannisg  it seems this enhancement should be done both on ARC and ARMv8. could I take ARMv8's enhancement ? Thanks.