zephyrproject-rtos/zephyr #15229: network tests have extremely restrictive whitelist

andrewboie, 2019-04-05:
I am seeing a great many network test cases that have the following:

```
common:
  depends_on: netif
  platform_whitelist: native_posix qemu_x86 qemu_cortex_m3
```

Why? We should be running these tests on any platform that has a network interface. We have no coverage at all on non-x86/non-ARM, and run nothing on hardware.

jukkar, 2019-04-06:
We had a lot more platforms earlier, but then the list was made much shorter in order to speed up sanitycheck time. Also we had lot of issues with different boards that have different requirements for memory consumption etc, but these can be overcome of course. I think this is more of a enhancement than a bug.

andrewboie, 2019-04-06:
I don't agree that this is an enhancement.
We have no coverage on RiscV32, Xtensa, ARC, Nios II.
We have no testing on hardware.
We have no testing on the ARM target which support stack overflow/userspace in emulation (mps2_an385)

> the list was made much shorter in order to speed up sanitycheck time

You do realize that if every subsystem did this, we would have hardly any testing at all? Why was it decided that network tests didn't need to run on every available platform, unlike everything else?

> Also we had lot of issues with different boards that have different requirements for memory consumption etc, but these can be overcome of course

These need to be worked through.

As it stands now, you've demoted this to 'future enhancement' which means it this is unlikely to get ever fixed...

jukkar, 2019-04-06:
> We have no coverage on RiscV32, Xtensa, ARC, Nios II.
> We have no testing on hardware.
> We have no testing on the ARM target which support stack overflow/userspace in emulation (mps2_an385)

That is a good list of additional boards, lets start with these to add more hw coverage to networking tests.

>  Why was it decided that network tests didn't need to run on every available platform, unlike everything else?

Most of the networking software is hw agnostic, they do not really depend on where they are run. But of course it is good to add more boards to the testing. I think we do not need to add for example all network enabled ARM boards to the testing, it just feels pointless. We could for example decide one specific board like sam-e70 or frdm-k64f for ARM board testing.

> As it stands now, you've demoted this to 'future enhancement' which means it this is unlikely to get ever fixed...

That was not my intention. I just marked this to "future" as this bug/enhancement will not go to 1.14 anyway and if I leave this setting empty someone will insists some number here. Also it is not clear in what release we will get more board coverage, but lets start to add more boards in 1.15 and continue from there.

andrewboie, 2019-04-06:
> lets start to add more boards in 1.15 and continue from there.

Works for me, thanks!

ioannisg, 2019-08-09:
@jukkar @andrewboie this issue is flagged as 2.0.0; is this taken care off via other issues/PRs?

andrewboie, 2019-08-09:
This is not yet resolved.
https://github.com/zephyrproject-rtos/zephyr/pull/17477 will close it.

dleach02, 2019-12-10:
> This is not yet resolved.
> #17477 will close it.

This was merged so I'm going to close this per comment from @andrewboie

closed 2019-12-10 by dleach02