zephyrproject-rtos/zephyr #15232: tests/bluetooth/tester fails build if CONFIG_TEST_USERSPACE=n

andrewboie, 2019-04-06:
Steps to reproduce:

```
diff --git a/tests/bluetooth/tester/default.conf b/tests/bluetooth/tester/default.conf
index 31dcfda0ad..b44411dd6d 100644
--- a/tests/bluetooth/tester/default.conf
+++ b/tests/bluetooth/tester/default.conf
@@ -34,3 +34,4 @@ CONFIG_BT_TESTING=y
 
 CONFIG_UART_CONSOLE=n
 CONFIG_GPIO=y
+CONFIG_TEST_USERSPACE=n
```

cmake output:

```
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD qemu_x86
-- Found west: /home/apboie/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading /home/apboie/projects/zephyr4/zephyr/boards/x86/qemu_x86/qemu_x86.dts as base
-- Overlaying /home/apboie/projects/zephyr4/zephyr/dts/common/common.dts
Parsing Kconfig tree in /home/apboie/projects/zephyr4/zephyr/Kconfig
Loading /home/apboie/projects/zephyr4/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig as base
Merging default.conf
Configuration written to '/home/apboie/projects/zephyr4/zephyr/tests/bluetooth/tester/out/zephyr/.config'

warning: CONSOLE_HANDLER (defined at drivers/console/Kconfig:38) was assigned the value 'y' but got
the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_CONSOLE_HANDLER.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.
-- Cache files will be written to: /home/apboie/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/i586-zephyr-elf/bin/i586-zephyr-elf-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module: tinycbor in path: /home/apboie/projects/zephyr4/modules/lib/tinycbor
CMake Warning at ../../../CMakeLists.txt:1468 (message):
  

        ------------------------------------------------------------
        --- WARNING:  __ASSERT() statements are globally ENABLED ---
        --- The kernel will run more slowly and use more memory  ---
        ------------------------------------------------------------

-- Configuring done
CMake Error at ../../../cmake/extensions.cmake:357 (add_library):
  No SOURCES given to target: drivers__gpio
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:334 (zephyr_library_named)
  ../../../drivers/gpio/CMakeLists.txt:1 (zephyr_library)

-- Build files have been written to: /home/apboie/projects/zephyr4/zephyr/tests/bluetooth/tester/out
```

andrewboie, 2019-04-06:
The CONFIG_GPIO=y in the default.conf seems to be causing this. Perhaps the error happens which a driver subsystem is selected, but no driver code at all to build? Userspace would cover this up since it would build the syscall handlers. @SebastianBoe any idea?

SebastianBoe, 2019-04-08:
> Perhaps the error happens which a driver subsystem is selected, but no driver code at all to build? Userspace would cover this up since it would build the syscall handlers.

That is correct. I would look into which driver sources should have been included, or look into if the driver shouldn't have been enabled in the first place.

PR #16306 (2019-05-22): tests/bluetooth/tester: remove spurious CONFIG_GPIO=y
This tries to build gpio without any driver which fails at cmake time
like this:

CMake Error at ../../../cmake/extensions.cmake:378 (add_library):
  No SOURCES given to target: drivers__gpio
  Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:355 (zephyr_library_named)
  ../../../drivers/gpio/CMakeLists.txt:3 (zephyr_library)

Remove TEST_USERSPACE workaround added in commit 3b53e6924958 which
added gpio_handlers.c to the SOURCES.

Fixes #15232.
commit 704b460b (2019-05-22): tests/bluetooth/tester: remove spurious CONFIG_GPIO=y

This tries to build gpio without any driver which fails at cmake time
like this:

CMake Error at ../../../cmake/extensions.cmake:378 (add_library):
  No SOURCES given to target: drivers__gpio
  Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:355 (zephyr_library_named)
  ../../../drivers/gpio/CMakeLists.txt:3 (zephyr_library)

Remove TEST_USERSPACE workaround added in commit 3b53e6924958 which
added gpio_handlers.c to the SOURCES.

Fixes #15232.
closed 2019-05-22 by nashif