zephyrproject-rtos/zephyr #15236: add external spi-nor flash will build fail

lgl88911, 2019-04-06:
**Describe the bug**
I add spi-nor flash for nrf52832 board, want to the external spi flash & nrf52832 internal flash work same times. but build error:
```
zephyr/drivers/flash/spi_nor.c:388:47: error: 'DT_JEDEC_SPI_NOR_0_ERASE_BLOCK_SIZE' undeclared here (not in a function)
zephyr/drivers/flash/spi_nor.c:409:22: error: 'DT_JEDEC_SPI_NOR_0_WRITE_BLOCK_SIZE' undeclared here (not in a function);
```

**To Reproduce**
Steps to reproduce the behavior:
1. modify my board dts, add spi-nor flash tree:
```
&spi2 {
	status = "ok";
	sck-pin = <4>;
	mosi-pin = <5>;
	miso-pin = <13>;

	cs-gpios = <&gpio0 15 0>;

	w25x16vsi0: w25x16vsi@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <2000000>;
		label = "W25X16VSI";
		jedec-id = <0xEF 0x80 0x3a>;
		size = <0x1000000>;
		erase-block-size = <0x10000>;
		write-block-size = <1>;
		hold-gpios = <&gpio0 9 0>;
	};
};
```
2. modify my board default conf，add spi-nor flash config
```
CONFIG_SPI_2=y
CONFIG_SPI_2_NRF_SPI=y
CONFIG_FLASH=y
CONFIG_SPI_NOR=y
CONFIG_SPI_NOR_CS_WAIT_DELAY=5
CONFIG_SPI_NOR_PAGE_SIZE=256
CONFIG_SPI_NOR_SECTOR_SIZE=4096
CONFIG_FLASH_HAS_PAGE_LAYOUT=y
CONFIG_FLASH_PAGE_LAYOUT=y
```
3. make, you can find undeclared DT_JEDEC_SPI_NOR_0_ERASE_BLOCK_SIZE & DT_JEDEC_SPI_NOR_0_WRITE_BLOCK_SIZE

**Expected behavior**
Build Pass & Spi-nor work well

**Additional context**
If the spi-flash be put in chosen, the DT_JEDEC_SPI_NOR_0_ERASE_BLOCK_SIZE & DT_JEDEC_SPI_NOR_0_WRITE_BLOCK_SIZE will be create
```
	chosen {
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,sram = &sram0;
		zephyr,flash = &w25x16vsi0;
		//zephyr,flash = &flash0;
	};
```
but, remove the internal flash0 will cause other build error

jimparis, 2019-04-08:
Hi,
I raised this same issue in PR https://github.com/zephyrproject-rtos/zephyr/pull/15225/
This patch should help: https://github.com/zephyrproject-rtos/zephyr/pull/15225/commits/cc5ffd03b6a5d258d7af8ccdf247402f017c0a00

lgl88911, 2019-04-09:
@jimparis  Thanks, I use your patch, build pass

PR #15225 (2019-04-11): SPI NOR driver fixes
Fix a typo, and define missing DT properties so that I can use spi-nor driver in addition to the default "chosen" flash driver for my platform (nrf52).

Fixes #15236
closed 2019-04-11 by carlescufi