zephyrproject-rtos/zephyr #15246: doc: confusion about dtc version

andrewboie, 2019-04-07:
I set up a workstation with Ubuntu 18.04 LTS and discovered that the device-tree-compiler is out of date:

```
CMake Error at /home/apboie/projects/zephyr/zephyr/cmake/extensions.cmake:1111 (message):
  Assertion failed: The detected dtc version is unsupported.

      The version was found to be 1.4.5
      But the minimum supported version is 1.4.6
      See https://docs.zephyrproject.org/latest/getting_started/
      for how to use the SDK's dtc alongside a custom toolchain.
```

However, the provided link gives no information about using an alternate dtc binary. I have the Zephyr SDK 0.10 set up.

andrewboie, 2019-04-07:
The SDK supposedly has a binary, but I can't seem to get the build to use it. If I do "sudo apt remove device-tree-compiler" it removes it, and then I get this when I try to build:

```
      The version was found to be 0.5.7
      But the minimum supported version is 1.4.6
```

I don't know where it is getting this 0.5.7 version of dtc

carlescufi, 2019-04-08:
>The SDK supposedly has a binary, but I can't seem to get the build to use it

Strange, this works perfectly well for me. I have DTC 1.4.5 but the build correctly uses the one in my SDK:

```
$ ~/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc --version
Version: DTC 1.4.7-g78e113e8-dirty
```

@andrewboie note that you need to set:
`ZEPHYR_SDK_INSTALL_DIR="/home/carles/bin/zephyr-sdk"`
for the build system to find your dtc.

andrewboie, 2019-04-08:
I have ZEPHYR_SDK_INSTALL_DIR set, but it still couldn't find it.
I had to download dtc manually compile and install it. Then I was able to run sanitycheck.

carlescufi, 2019-04-08:
@andrewboie I am puzzled then because it should search first `CMAKE_PREFIX_PATH` and then `PATH`. And it is being added to `CMAKE_PREFIX_PATH` according to:
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/toolchain/zephyr/0.10.0/host-tools.cmake#L9

Can you please double-check that your `ZEPHYR_SDK_INSTALL_DIR` is pointing to a correct installation of 0.10.0?

galak, 2019-04-09:
@andrewboie do you happen to have DTC set in your environment?

carlescufi, 2019-04-09:
> @andrewboie do you happen to have DTC set in your environment?

I thought about that but even then `find_program` should give precedence to the `CMAKE_PREFIX_PATH` according to CMake doc.

andrewboie, 2019-04-09:
This must have something weird but transitory in my environment when I was setting up this workstation, seems to be fine now. Sorry for the noise!

dbkinder, 2019-05-24:
@carlescufi @andrewboie FYI, I'm seeing the same misbehavior (not finding the dtc in the Zephyr SDK).  I also observed a new user following the getting started instructions on a clean Ubuntu system and having the same problem.

For this to work properly you need to set **two** environment variables:

         export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
         export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk

I had misspelled ZEPHYR_TOOCHAIN_VARIANT (left off the L), doh.  For the "new user" they only set the ZEPHYR_SDK_INSTALL_DIR variable and not ZEPHYR_TOOLCHAIN_VARIANT.

tsaitsai, 2019-07-05:
Hi,
I'm running into the same issue, also on a new Ubuntu 18 installation.

> ~/zephyrproject/zephyr/samples/hello_world$ west build -b nrf51_pca10031-- west build: build configuration:
>        source directory: /home/hp1/zephyrproject/zephyr/samples/hello_world
>        build directory: /home/hp1/zephyrproject/zephyr/samples/hello_world/build
>        BOARD: nrf51_pca10031 (origin: command line)
> -- west build: generating a build system
> Zephyr version: 1.14.99
> -- Selected BOARD nrf51_pca10031
> -- Found west: /home/hp1/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
> CMake Error at /home/hp1/zephyrproject/zephyr/cmake/extensions.cmake:1324 (message):
>   Assertion failed: The detected dtc version is unsupported.
> 
>       The version was found to be 1.4.5                                   
>       But the minimum supported version is 1.4.6                        
>       See https://docs.zephyrproject.org/latest/getting_started/                     
>       for how to use the SDK's dtc alongside a custom toolchain.
> Call Stack (most recent call first):
>   /home/hp1/zephyrproject/zephyr/cmake/host-tools.cmake:69 (assert)
>   /home/hp1/zephyrproject/zephyr/cmake/app/boilerplate.cmake:412 (include)
>   CMakeLists.txt:5 (include)
> 
> -- Configuring incomplete, errors occurred!

I tried adjusting the two enviornmental variables as @dbkinder mentioned, but still did not fix it.  In my case, my sdk directory is inside my home directory.

`export ZEPHYR_TOOLCHAIN_VARIANT=zephyr`
`export ZEPHYR_SDK_INSTALL_DIR=$HOME/zephyr-sdk-0.10.1`

andrewboie, 2019-07-26:
I'm reopening this.
People are still reporting this issue on the mailing list.

yenWu, 2019-08-01:
Same problem here. I tried to build the sample(hello world) on RISCV qemu. But the DTC:FILEPATH link to the local `dtc`. 
```
//zephyr/build-example/CMakeCache.txt
...
DTC:FILEPATH=/usr/local/dtc
```
So I manually replaced it with `/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc`, and it worked. But I failed to reproduce it. It seems there might be an unexpected condition which makes it link to local `dtc` in cmake files.

carlescufi, 2019-08-15:
This typically happens only if you already have a build folder that has found a system-wide DT. The solution is to completely remove the build folder and run CMake again. This has been confirmed several times, so closing this unless @yenWu still sees the issue even after removing the full build folder.

closed 2019-08-15 by carlescufi

ntavish, 2019-10-30:
Issue is closed but just as an update, I think whoever is seeing this issue must have moved the SDK location after installation.

For example on my system, the `dtc` path right now is `/opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc`. On trying to execute the file I just see:
```
tavish@resurgam:~ $ /opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
bash: /opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: No such file or directory
tavish@resurgam:~ $ file /opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
/opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /opt/zep, for GNU/Linux 3.2.0, BuildID[sha1]=9a0e353686d13fed7e6496f553c3838571239bd2, stripped
```

The dynamic linker/loader path is different than it should be because I must have moved it after installation:

```
tavish@resurgam:~ $ readelf -a /opt/zephyr-sdk-0.10/sysroots/x86_64-pokysdk-linux/usr/bin/dtc | grep interp
  [ 1] .interp           PROGBITS         0000000000400238  00000238
      [Requesting program interpreter: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2]
```

maksimmasalski, 2020-08-24:
I have the same problem on Ubuntu 18.04.4 LTS. Solution here and deleting build/ folder didn't help me. https://github.com/zephyrproject-rtos/zephyr/issues/17543