zephyrproject-rtos/zephyr #15247: tests/posix/common: test_posix_pthread_execution() reboots repeatedly for portability.posix

cinlyooi-intel, 2019-04-08:
**Describe the bug**
test_posix_phread_execution()reboots repeatedly.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make
4. make run

**Screenshots or console output**
```
***** ***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
[... 204 lines cut ...]
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-148-g318a98a7fb1f (delayed boot 1000ms) *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
***** delaying boot 1000ms (per build configuration) *****
```

**Environment (please complete the following information):**
 - OS: fedora
 - Toolchain : zephyr
 - Commit SHA or Version used: 318a98a7fb1f604369ff1f2ee99608836176a0d4

andrewboie, 2019-04-09:
that test sits for a few seconds after those printouts (it's doing a lot of context switch thrashing between threads) and then continues...could this be a watchdog reset?

MaureenHelm, 2019-04-10:
I don't think this is a watchdog reset. The watchdog gets disabled during boot and doesn't get re-enabled unless you enable the watchdog driver (which we don't by default).

We are getting a usage fault and git bisect is pointing to:
```
commit 7df0216d1e2a2800225d2d3756f5decb0568188d
Author: Andy Ross <andrew.j.ross@intel.com>
Date:   Wed Jul 25 13:55:59 2018 -0700

    kernel/mutex: Spinlockify
    
    Use a subsystem lock, not a per-object lock.  Really we want to lock
    at mutex granularity where possible, but (1) that has non-trivial
    memory overhead vs. e.g. directly spinning on the mutex state and (2)
    the locking in a few places was originally designed to protect access
    to the mutex *owner* priority, which is not 1:1 with a single mutex.
    
    Basically the priority-inheriting mutex code will need some rework
    before it works as a fine-grained locking abstraction in SMP.
    
    Note that this fixes an invisible bug: with the older code,
    k_mutex_unlock() would actually call irq_unlock() twice along the path
    where there was a new owner, which is benign on existing architectures
    (so long as the key argument is unchanged) but was never guaranteed to
    work.  With a spinlock, unlocking an unlocked/unowned lock is a
    detectable assertion condition.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>
```
cc: @andyross

MaureenHelm, 2019-04-12:
I think this is related to #15411. The same suspicious commit 2895da02a44980e901fbcdd7ec87edc2016605c5 changes this test from failing to passing!

andrewboie, 2019-04-12:
@MaureenHelm I think you may be onto something. For unrelated reasons I was looking into the implementations of tests/kernel/critical and tests/posix/common for #14642 . These two tests have something in common: they both do a crazy amount of context switch thrashing between various threads.

So if we have some kind of race condition involved with context switching, it's reasonable for both these tests to encounter problems.

nashif, 2019-04-12:
this bug however was opened before 2895da0 was committed...

MaureenHelm, 2019-04-12:
> this bug however was opened before [2895da0](https://github.com/zephyrproject-rtos/zephyr/commit/2895da02a44980e901fbcdd7ec87edc2016605c5) was committed...

The test fails before that commit and passes after it.

nashif, 2019-04-12:
> The test fails before that commit and passes after it.

oh, I misread the comment above

ioannisg, 2019-04-12:
This makes me think the root-cause of these fails might be on kernel - not on the timer driver side.

nashif, 2019-04-12:
the test fails if you add CONFIG_BOOT_DELAY=1000

andyross, 2019-04-12:
This board has a watchdog, right?  Is it just that the boot delay is happening with interrupts locked and/or in a state where the watchdog hasn't been set up for the right timeout yet?

MaureenHelm, 2019-04-12:
The watchdog gets disabled very early in the boot process before the delay, and doesn't get reenabled unless `CONFIG_WATCHDOG=y`. I'm also able to reproduce this without the boot delay.

PR #15469 (2019-04-16): arch: arm: mpu: Lock irqs while reprogramming the nxp mpu
The nxp mpu uses the logical OR of access permissions when multiple
region descriptors apply to a given memory access. This means that we
must partition the sram into two non-overlapping regions to implement
the mpu stack guard. This partitioning gets reconfigured at every
context switch, and if an interrupt occurs during this time, it can
cause a fault because we do not have a valid mpu region descriptor for
the sram.

This scenario was observed on frdm_k64f in tests/posix/common before
commit 2895da02a44980e901fbcdd7ec87edc2016605c5,  [... cut]
closed 2019-04-16 by nashif