zephyrproject-rtos/zephyr #15251: nRF Watchdog not triggering on kernel panic

tautologyclub, 2019-04-08:
If I configure watchdog and try it out with for example

```
while(1);
```

a wdt reset is always triggered. But if I try to trigger it with

```
k_panic();
```

the watchdog does nothing. The system just hangs indefinitely.

tautologyclub, 2019-04-08:
Actually, the following snippet doesn't trigger it either:

    typedef void (*bla)(void);
    ...
        	/* Try pissing off the MPU */
        	bla null_func = (bla)0;
	        null_func();

carlescufi, 2019-04-08:
@nordic-krch @kl-cruz @andrewboie what's the expected behavior for watchdogs when a kernel panic is triggered?

ioannisg, 2019-04-08:
the system is expected to hang:

```
	if (reason == _NANO_ERR_KERNEL_PANIC) {
		goto hang_system;
	}
```

ioannisg, 2019-04-08:
> 
> 
> Actually, the following snippet doesn't trigger it either:
> 
> ```
> typedef void (*bla)(void);
> ...
>     	/* Try pissing off the MPU */
>     	bla null_func = (bla)0;
>         null_func();
> ```

Can you post the exact output you see?

andrewboie, 2019-04-08:
> The system just hangs indefinitely.

What are you expecting to happen when k_panic() is called?
This does not seem like a bug to me.

closed 2019-04-08 by andrewboie

henrikbrixandersen, 2019-04-08:
I would at least expect to be able to configure this behavior. A system watchdog is not of much use, if not for resetting the CPU when an error occurs.

anangl, 2019-04-09:
@henrikbrixandersen 
> I would at least expect to be able to configure this behavior. A system watchdog is not of much use, if not for resetting the CPU when an error occurs.

Just configure and enable the watchdog then. When its timer expires, it will reset the CPU, no matter if 
it is hung (in `k_panic();`, `while (1);`, or any other infinite loop), or it runs normally. That's how watchdogs work. They are to reset the system when they are not fed (their timers are not reset) for a certain (mostly configurable) time. But they need to be permitted to do so (enabled).

@tautologyclub 
> But if I try to trigger it with
> 
> ```
> k_panic();
> ```
> the watchdog does nothing. The system just hangs indefinitely.

I am not sure what you mean by "if I try to trigger it". You don't trigger a watchdog. You can only enable it, and then the watchdog timer must expire for the configured further action (e.g. reset) to be triggered. Have you enabled the watchdog before calling `k_panic()`?

carlescufi, 2019-04-09:
@andrewboie @ioannisg @tautologyclub maybe `k_panic()` should reboot?

ioannisg, 2019-04-09:
So, first, let me present some findings. It looks to me that the watchdog (for nRF) works as expected. Here's my modified hello-world (inspired by the watchdog test):

```
#include <zephyr.h>
#include <misc/printk.h>
#include <watchdog.h>

void main(void)
{
	int err;
	struct wdt_timeout_cfg m_cfg_wdt0;

	printk("Hello World! %s\n", CONFIG_BOARD);

	struct device *wdt = device_get_binding(DT_WDT_0_NAME);

	if (!wdt) {
		printk("Cannot get WDT device\n");
		return;
	}

	m_cfg_wdt0.callback = NULL;
	m_cfg_wdt0.flags = WDT_FLAG_RESET_SOC;
	m_cfg_wdt0.window.max = 2000U;
	err = wdt_install_timeout(wdt, &m_cfg_wdt0);
	if (err < 0) {
		printk("Watchdog install error\n");
		return;
	}

	err = wdt_setup(wdt, 0);
	if (err < 0) {
		printk("Watchdog setup error\n");
		return;
	}

	k_panic();
}
```

Here's the output I get (goes forever):

```
Hello World! nrf52840_pca10056
***** Kernel Panic! *****
Current thread ID = 0x200000f0
Faulting instruction address = 0x4fe
Hello World! nrf52840_pca10056
***** Kernel Panic! *****
Current thread ID = 0x200000f0
Faulting instruction address = 0x4fe
Hello World! nrf52840_pca10056
***** Kernel Panic! *****
Current thread ID = 0x200000f0
Faulting instruction address = 0x4fe
Hello World! nrf52840_pca10056
***** Kernel Panic! *****
Current thread ID = 0x200000f0
Faulting instruction address = 0x4fe
Hello World! nrf52840_pca10056
***** Kernel Panic! *****
Current thread ID = 0x200000f0
Faulting instruction address = 0x4fe

```

So it seems to me that the kernel is in panic, but the watchdog still reboots the MCU at expire time.

ioannisg, 2019-04-09:
> 
> 
> @andrewboie @ioannisg @tautologyclub maybe `k_panic()` should be configurable to either hang indefinitely or reboot?

So, k_panic() triggers the System Fatal Error Handler, which, in /arch has a weak implementation

`void __weak z_SysFatalErrorHandler(unsigned int reason, const NANO_ESF *pEsf)`

and simply hangs the CPU, but does not stop/reset the watchdog, or disables interrupts, AFAICS, so the behavior we see is the expected one.

I don't think we need to enforce system reboot on k_panic(); this is something the app developer can do by overwriting the SysFatalErrorHandler.

tautologyclub, 2019-04-09:
@ioannisg Of course I configured the watchdog before I did `k_panic()`. I expected the watchdog reset to trigger, but it didn't. It is only triggered by infinite busy waiting. My code is pretty much identical to yours w.r.t. setup. The only difference is that I used the flags

```(WDT_OPT_PAUSE_IN_SLEEP | \
			  WDT_OPT_PAUSE_HALTED_BY_DBG)```

Actually just now I tried commenting out `k_cpu_idle();` inside the infinite loop inside `z_SysFatalErrorHandler` just now. And that actually made it so kernel panic triggered the watchdog. So it seems that `k_cpu_idle();` causes the watchdog not to reset when `WDT_OPT_PAUSE_IN_SLEEP` is set. Is this a bug, or expected behaviour? Well I dunno, but it was certainly not what I had expected.

tautologyclub, 2019-04-09:
@andrewboie You sure this issue is closed?