zephyrproject-rtos/zephyr #15256: Link Layer Control Procedure sequencing

carlescufi, 2019-04-08:
There are two types of issues that can occur today when handling Control Procedures in the Link layer:

* Type 1: Peer initiated procedure first and then local initiated procedure will lead to `HCI_CMD_DISALLOWED` 
* Type 2: Local initiated procedure first and then peer initiated procedure will lead to disconnect

We need 2 tables, one per Type, listing the combinations of Control Procedures that can lead to the undesired outcome (`HCI_CMD_DISALLOWED` or disconnection.

## Type 1

## Type 2

joerchan, 2019-07-09:
Attached sniffer log of type 3
[type3.zip](https://github.com/zephyrproject-rtos/zephyr/files/3371892/type3.zip)

maz3max, 2019-07-23:
A sniffer log where both peers try to set the security level to FIPS.
[both_check.zip](https://github.com/zephyrproject-rtos/zephyr/files/3422739/both_check.zip)

maz3max, 2019-07-23:
A sniffer log where only the central tries to set the security level to FIPS (and fails).
[only-central-checks.zip](https://github.com/zephyrproject-rtos/zephyr/files/3422752/only-central-checks.zip)

maz3max, 2019-08-25:
@joerchan After the merge of #17836, the peripheral still cannot set the security level right after connecting. I suspect the problem to be on the central side this time.
Logs are here: https://pastebin.com/55kritYE
Sniffer Capture is here (it loops a few times):
[only-periph-req-sec.zip](https://github.com/zephyrproject-rtos/zephyr/files/3538111/only-periph-req-sec.zip)

carlescufi, 2019-08-25:
> @joerchan After the merge of #17836, the peripheral still cannot set the security level right after connecting. I suspect the problem to be on the central side this time.
> Logs are here: [...]

Thanks for the report @maz3max. @joerchan can you decide whether this requires a new GH issue to be open, or will it be fixed by an existing PR?

joerchan, 2019-08-26:
@carlescufi @maz3max :
Reproduced the issue and reported here: https://github.com/zephyrproject-rtos/zephyr/issues/18645

carlescufi, 2019-08-30:
@pskhansen @cvinayak please discuss the resolution of this issue, there are two options:

- Merge #15320, port it to the split link layer
- Come up with a new approach for control procedures for the split LL, merge #15320 for the legacy link layer

carlescufi, 2020-07-06:
@kruithofa @erbr-ot @thoh-ot this is the issue tracking the LLCP rework.

kruithofa, 2020-07-08:
We should also have a look at issue #13519

fabiobaltieri, 2022-08-11:
@carlescufi can you update this one? Looks like all issues/PRs linked here have been merged/closed, is this done?

carlescufi, 2022-08-17:
> @carlescufi can you update this one? Looks like all issues/PRs linked here have been merged/closed, is this done?

This can be considered as done for 3.2.0 @fabiobaltieri

closed 2022-09-09 by mmahadevan108