zephyrproject-rtos/zephyr #15259: CAN sample does not work

dykon, 2019-04-08:
CAN sample does not work

We try to get the ```zephyr\samples\net\sockets\can``` sample to work on a ```nucleo_l432```

Compilation and flashing works without problem.

Serial output is:

```
***** Booting Zephyr OS v1.14.0-rc3-152-gbfb7e4531c91 *****
***** Reserved Exception ( -16) *****
***** Hardware exception *****
Current thread ID = 0x20000cec
Faulting instruction address = 0x6f742045
Fatal fault in essential thread! Spinning...
```

alexanderwachter, 2019-04-08:
@ dykon what does your setup looks like? Do you have a CAN transceiver attached to the board?
Can you enable Asserts and CAN drivers debug log and try it again?

jukkar, 2019-04-09:
Just tried this with `nucleo_l432kc` board and I did not see any errors.

alexanderwachter, 2019-04-09:
@jukkar @dykon I guess that the CAN controller is not ready or something like that. We will see this in the logs.
@jukkar did you tried socketCAN without an initialized CAN controller?

jukkar, 2019-04-09:
> did you tried socketCAN without an initialized CAN controller?

Note sure what you mean by that. I just flashed the socket_can sample to two nucleo_l432kc boards. Then monitored the 1st board that it is not crashing. After that I started 2nd board which was connected to 1st board. Monitored both boards and did not see any errors / crashes. So this setup worked for me.

alexanderwachter, 2019-04-09:
I just wonder why it crashes. Maybe it is because the controller initialization fails (for example the CAN_RX pin is not high when the driver is initializing).

markusr, 2019-04-13:
I also want to use CAN on ```nucleo_l432```. There is no additional CAN hardware attached.

I added to both  ```proj.conf```
```
CONFIG_CAN_LOG_LEVEL_DBG=y
CONFIG_PRINTK=y
CONFIG_LOG=y
CONFIG_TEST=y
```

For  ```samples/net/sockets/can``` the output is

```
ASSERTION FAIL [net_if_get_link_addr(iface)->addr != ((void *)0)] @ /home/markus/zephyr/west/zephyr/subsys/net/ip/net_if.c:2867
***** Kernel Panic! *****
Current thread ID = 0x20000d10
Faulting instruction address = 0x800cbbe
```

For  ```samples/drivers/CAN``` the output is:
```
***** Booting Zephyr OS v1.14.0-rc3-245-gd1176d1789e2 *****
CAN: Device driver not found.
[00:00:00.020,000] <err> stm32_can: HAL_CAN_Init failed: 3
```

It seems that ```HAL_CAN_Init```returns a timeout.

Maybe there are 2 problems:
- can-driver on stm32_l4 series does not work
- socketcan fails if can-driver fails

alexanderwachter, 2019-04-13:
@markusr the CAN controller fails during initialization because you don't have a CAN transceiver attached. The controller only initializes correctly if there are at least 11 cycles recessive state (high value at the CAN_RX pin) during initialization.

jukkar, 2019-04-15:
This is basically hw configuration/setup issue, meaning that the transceiver is missing -> closing.

closed 2019-04-15 by jukkar

alexanderwachter, 2019-04-15:
@dykon @markusr you can write me on slack if you have any further questions.