zephyrproject-rtos/zephyr #15286: HF clock's m16src_grd and BLE stack

pdunaj, 2019-04-09:
**Describe the bug**
I was reviewing HF start and stop in `nrf_power_clock.c` and noticed that these functions use `m16src_grd`. The idea is to unlock the interrupts but mark the clock operation as pending until in finishes. While the operation is pending a concurrent call to `m16src_start` or `m16src_stop` will return with `-EAGAIN`. The problem is BLE stack is ignoring all return values from HF start/stop.

The following problem could happen while start/stop are called concurrently:
- BLE enable clock while it is being disabled by the other party.
The `m16src_ref` is decremented to zero by the other party and `m16src_grd` is set. CPU is taken from the other party just before executing `m16src_grd = 0`. BLE tries enabling the clock but will get `-EAGAIN` which will be ignored. HF clock will remain disabled and transmission will not occur.
- BLE disable the clock after first case happened and the other party re-enabled the clock on its side. 
The other party re-enables the clock - ref count is 1. BLE does not know clock start failed (error is not checked) and tries to stop it. Stop will decrements ref count and cause HF clock to be disabled even if it is used by the other party.

The problem is a race and will depend on the priorities used.

**To Reproduce**
N/A

**Expected behavior**
No races.

**Impact**
Unknown

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
present since `5843e8df898`, still there in `7eca256`

**Additional context**
N/A

pdunaj, 2019-04-09:
Ah... and there is no barrier before releasing the guard.

carlescufi, 2019-08-26:
@pdunaj @anangl @cvinayak is this still an issue?

pdunaj, 2019-08-27:
@carlescufi possibly yes. I had a brief look in controller files. Code touching the clock in `subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c`  is checking the error code now. I have not checked if this is properly handled in upper layers.

carlescufi, 2020-06-03:
@cvinayak @pdunaj @nordic-krch is this still applicable?

carlescufi, 2020-09-14:
@cvinayak @pdunaj @nordic-krch is this still applicable?
If not, please close

pdunaj, 2020-09-14:
As far as I know this should be fixed by using onoff to control the operation of hfclock from multiple sources.

closed 2020-09-14 by pdunaj