zephyrproject-rtos/zephyr #15299: west init fails in powershell

E3V3A, 2019-04-09:
I'm following your west instructions from [here](https://docs.zephyrproject.org/latest/getting_started/index.html) with the result that `west init` fails, like this:

```bash
$ west --version
West bootstrapper version: v0.5.7 (c:\python37\lib\site-packages\west\_bootstrap)

$ west init zephyrproject
Initializing in new directory zephyrproject
←[92m=== Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr, rev. master into C:\NordicS\ncs\zephyrproject\.west\tmp
←[0mTraceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python37\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 481, in main
    init(wrap_argv[1:])
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 244, in init
    bootstrap(args)
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 310, in bootstrap
    exist_ok=True)
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 114, in clone
    subprocess.check_call(('git', 'init', dest))
  File "c:\python37\lib\subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "c:\python37\lib\subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "c:\python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "c:\python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

With PWSH version:
```powershell
$ $PSVersionTable

Name                           Value
----                           -----
PSVersion                      6.1.1
PSEdition                      Core
GitCommitId                    6.1.1
OS                             Microsoft Windows 6.3.9600
Platform                       Win32NT
PSCompatibleVersions           {1.0, 2.0, 3.0, 4.0...}
PSRemotingProtocolVersion      2.3
SerializationVersion           1.1.0.1
WSManStackVersion              3.0
```

mbolivar, 2019-04-09:
Hi @E3V3A, and thanks very much for the report.

Zephyr in general is tested using cmd.exe on Windows. Powershell support is neither expected nor guaranteed. Are you able to work with cmd.exe (or the Bash that comes with git)? I've tested west with both of those today on Windows 10 and had no issues.

E3V3A, 2019-04-09:
@mbolivar 
Hi. Thanks for quick feedback. I'm just trying to get my SEGGER ARM Development kit to work, and it seem to require west. (As I explained in detail [here](https://github.com/zephyrproject-rtos/zephyr/issues/15293). )

> Are you able to work with cmd.exe (or the Bash that comes with git)?

I can try to use whatever, as long as it's not WSL, since I'm on W8.1. I tried normal bash, but that resulted in a very similar error. But I need to know if there are additional settings required. E.g.

* What are the correct environment variables to be set?
* Do I need to setup CMake?
* What are the commands you'd suggest me to try?

mbolivar, 2019-04-09:
> I can try to use whatever, as long as it's not WSL, since I'm on W8.1. I tried normal bash, but that resulted in a very similar error. But I need to know if there are additional settings required. 

The details are all covered in the getting started guide: https://docs.zephyrproject.org/latest/getting_started/index.html

West doesn't require any additional environment variables.

The windows specific document inside has a `cmd.exe` setup option: https://docs.zephyrproject.org/latest/getting_started/installation_win.html

WSL is also covered there; you can ignore it, it's just another option.

> BTW. It seem that west init tries to use .west\tmp instead of what docs say: .west\west!

That's where zephyr gets cloned to temporarily during `west init`. It will be moved to the right place if `west init` finishes successfully.

E3V3A, 2019-04-09:
@mbolivar 

Come on guys, ^^ that is not very helpful! (Referring back to the same document I already mentioned I was following.)

> West doesn't require any additional environment variables.

So what is [`this`](https://github.com/zephyrproject-rtos/zephyr/blob/master/zephyr-env.cmd#L2) used for? (When is it supposed to be used?)

`set ZEPHYR_BASE=%~dp0`

> The windows specific document inside has a cmd.exe setup option

No it doesn't, did you actually read it?

Have you actually tried to walk through your own installation instructions on a fresh windows install? 
I find that hard to believe...but would love to see your log how you did it. :+1:

carlescufi, 2019-04-09:
@E3V3A I recently went through a full setup of west on a clean machine using Powershell without any issues. I was helping another user getting set up. Can you please verify that you can reach `git` from your powershell prompt? I suspect that is the issue. You need to add the folder containing `git.exe` to your `PATH`. Could you try that and let us know?

carlescufi, 2019-04-09:
>So what is this used for? (When is it supposed to be used?)

We need to clarify that. It is required when building using CMake and ninja but not when building using west.

>> The windows specific document inside has a cmd.exe setup option
> No it doesn't, did you actually read it?

https://docs.zephyrproject.org/latest/getting_started/installation_win.html#option-1-windows-command-prompt

It mentions `cmd.exe` by name doesn't it?

carlescufi, 2019-04-09:
>What are the correct environment variables to be set?

Assuming you are building for Arm:
* `ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb`
* `GNUARMEMB_TOOLCHAIN_PATH=c:\path\to\gnu\arm\embedded`

See https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded

>Do I need to setup CMake?

You need install CMake, yes. You can use Chocolatey as described here:
https://docs.zephyrproject.org/latest/getting_started/installation_win.html#option-1-windows-command-prompt

>What are the commands you'd suggest me to try?

Add the folder where your `git.exe` lives to your `PATH`, then run `west init zephyrproject` on a clean folder

mbolivar, 2019-04-09:
> So what is this used for? (When is it supposed to be used?)
> set ZEPHYR_BASE=%~dp0

Sorry, I meant to say that west doesn't require any additional environment variables beyond what's in the getting started. In fact, we're trying to decrease the use of mandatory environment variables in general by making them west configuration options. ZEPHYR_BASE is an example; you don't need to set it if you're using west to build, flash, and debug. You still need to set the other variables covered in the documentation, which Carles has also listed in this thread.

>> The windows specific document inside has a cmd.exe setup option
> No it doesn't, did you actually read it?

Just for reference, in case it is hard to find, this is what I'm referring to:

[image]

> Have you actually tried to walk through your own installation instructions on a fresh windows install?

I just did that today, coincidentally, both in cmd.exe and in Git's Bash shell on Windows 10. The test I mentioned in my first comment was on a fresh machine.

Sounds like Carles has a theory about what's happening on your computer.

I know Zephyr can be a bit hard to set up. We are working on making it easier. A first step was adding commands like `west build` etc. so you don't always have to set ZEPHYR_BASE and do multiple `cmake`/`ninja` lines. Thanks for your feedback; we'll try to keep improving.

E3V3A, 2019-04-09:
Hi Guys! Good News!

The culprit *seem* to have been a combination of pwsh 6.1 (when using chocolatey install) and indeed the **git.exe** culprit. Apparently the verification steps for git passed, making me believe it was working. However, something had picked up the wrong git. (From Cygwin, which we know is partially broken for native Windows use, due to a known bug but lacking updates to its Cygwin package.) 

Re-installing git with a working Chocolatey fixed the issue. (`choco install git`)

I'll close this.  Thanks again for your patience.

closed 2019-04-09 by E3V3A

carlescufi, 2019-04-09:
Thanks for the feedback and for following-up @E3V3A. Know that we have just merged a PR into the west repo improving the error code when Git is not found after you  brought it up.

mbolivar, 2019-04-09:
> Re-installing git with a working Chocolatey fixed the issue. (choco install git)

Great! Thanks, glad to hear it.