zephyrproject-rtos/zephyr #15315: doc: simplify cmake examples thanks to cmake's new -B option

marc-hb, 2019-04-10:
This issue is to request that the cmake examples in the documentation are simplified thanks to the new -B option.

CMake versions before 3.13 required running cmake from the temporary `build` directory, example:

https://docs.zephyrproject.org/1.13.0/getting_started/getting_started.html#building-a-sample-application

As explained in the CMake FAQ, one of the major advantages of out-of-source builds is that cleaning is as simple as throwing away the `build` directory:
https://gitlab.kitware.com/cmake/community/wikis/FAQ#cmake-does-not-generate-a-make-distclean-target-why
> A "make distclean" target is only necessary if the user performs an in-source build. CMake supports in-source builds, but we strongly encourage users to adopt the notion of an out-of-source build. [...] Because CMake does not change the source tree, there is no need for a distclean target. One can start a fresh build by deleting the build tree or creating a separate build tree.

However deleting the current directory is obviously not recommended on Unix and probably not possible on Windows. So starting over requires all these commands:
```
cd ..
rm -rf build/
mkdir build
cd build
cmake -DBOARD=some_board ..
make
```
Very likely realizing this usability issue, cmake 3.13 (required by Zephyr since PR #12022) introduced a new -B option which completely removes the need to change directory and simplifies all the commands above to just these:
```
rm -rf build/
cmake -B build -DBOARD=some_board
make -C build
```

This issue is to request that the cmake examples in the documentation are simplified thanks to this new -B option. As these examples are generated, this requires updating the script that was implemented in PR #13279

aescolar, 2019-04-10:
CC @SebastianBoe @dbkinder

SebastianBoe, 2019-04-10:
I agree that we should update the documentation to use the -B flag.

marc-hb, 2019-04-17:
I think this doc change would be especially useful considering how CMake's cache seems to silently discard most user changes of mind... :-( Deleting build/ has become second nature for me.

andrewboie, 2019-04-17:
> starting over requires all these commands:

```
cd ..
rm -rf build/
mkdir build
cd build
cmake -DBOARD=some_board ..
make
```

Can't you just do:

```
rm -rf *
cmake -DBOARD=some_board ..
make
```

marc-hb, 2019-04-17:
> rm -rf *

Two issues:
- doesn't delete dot (hidden) files
- This command is too dangerous. At the very least I would do rm -rf ../build/*

andrewboie, 2019-04-17:
> doesn't delete dot (hidden) files

There aren't any unless I'm missing something:

```
apboie@yggdrasil:~/projects/zephyr3/zephyr/samples/hello_world/out master /home/apboie/projects/zephyr3/zephyr
$ ls -lah
total 80K
drwxr-xr-x  6 apboie apboie 4.0K Apr 17 11:18 .
drwxr-xr-x  4 apboie apboie 4.0K Apr 17 11:17 ..
drwxr-xr-x  2 apboie apboie 4.0K Apr 17 11:18 app
-rw-r--r--  1 apboie apboie  19K Apr 17 11:18 CMakeCache.txt
drwxr-xr-x  9 apboie apboie 4.0K Apr 17 11:18 CMakeFiles
-rw-r--r--  1 apboie apboie 1.6K Apr 17 11:18 cmake_install.cmake
-rw-r--r--  1 apboie apboie   79 Apr 17 11:18 Kconfig.modules
-rw-r--r--  1 apboie apboie  22K Apr 17 11:18 Makefile
drwxr-xr-x  3 apboie apboie 4.0K Apr 17 11:18 tinycbor
drwxr-xr-x 15 apboie apboie 4.0K Apr 17 11:18 zephyr
-rw-r--r--  1 apboie apboie   64 Apr 17 11:18 zephyr_modules.txt
```

> This command is too dangerous. At the very least I would do rm -rf ../build/*

Seems a matter of opinion.

marc-hb, 2019-04-17:
> There aren't any [dot files]

... until the next CMake/west/ninja/zephyr/whatever version. Too low-level implementation details; not interested in debugging potential issues like this - can be very time-consuming.

> Seems a matter of opinion.

I did lose data a few times by running similar commands in the wrong directory or by picking them up by mistake when re-using my history too fast and I did hear and read similar stories. Maybe we're just a minority of clumsy people (including the persons behind the new -B option?) but for sure painful facts and not just opinion. 

Sometimes I even manually `mv stuff ~/Trash/`.

The suggested change also saves the `../` in `edit ../src/foo.c`, will edit the description.

PS @carlescufi : considering Windows is obviously more "GUI-oriented", I'm curious about the equivalent cmd.exe / Powershell "experience" if any wrt to this general cmake cleanup question.

carlescufi, 2019-04-18:
Several things to comment on here:

- `ninja pristine` cleans up everything but the build folder itself, so we could use that  instead of `rm`, avoiding differences between NIX and Windows
- If, in the examples in the doc,  we assume the user is inside the tree, we can also use `git clean -fdx`
- Windows uses `rmdir /Q /S`, which is quite ugly and hard to type
- We just merged a PR into west that allows you to do: `west -p -b <board> <sample>` and that runs the `ninja pristine` target before building

So I am unsure about whether we want to pursue documenting this in detail or just switch to using west in the doc, which is easier and cross-platform.

@mbolivar thoughts?

mbolivar, 2019-04-18:
> @mbolivar thoughts?

Not sure why we're on the subject of how to delete files even after reading these comments. The new `-B` flag is an improvement no matter how you look at it AFAICT, even on a fresh build:

```
mkdir build
cd build
cmake -GNinja -DBOARD=myboard ..
ninja
```

vs:

```
cmake -Bbuild -GNinja -DBOARD=nrf52840_pca10056 samples/hello_world/
ninja -C build
```

Does anybody disagree we should use `-B`? If so, why?

marc-hb, 2019-04-18:
> ninja pristine cleans up everything but the build folder itself, so we could use that instead of rm, avoiding differences between NIX and Windows

Sorry for the digression and confusion; as noted by @mbolivar this issue is _not_ about documenting how to clean up. Cleaning up build directories was just one example. Another example is building multiple configurations in different build directories (without cleaning up)

> So I am unsure about whether we want to pursue documenting this in detail or just switch to using west in the doc, 

Ignoring what would be the best documentation for a minute, "just switching" cmake examples to west would be a much bigger documentation change and task than merely tweaking the script that generates cmake examples. 

> which is easier and cross-platform.

cmake is cross-platform too (unlike `rm -rf`, sorry again for the digression)

@carlescufi you may be right; if you feel like it please file *another*, orthogonal issue for "less cmake and more west examples in the doc". \<OFF TOPIC\>I wouldn't aim for "zero cmake examples" though because developers are not consumers and the documentation shouldn't try to keep them totally ignorant of what's happening behind the scenes (blah "[…except for the problem of too many levels of indirection](https://en.wikipedia.org/wiki/Fundamental_theorem_of_software_engineering)" blah)

marc-hb, 2019-06-15:
>  if you feel like it please file another, orthogonal issue for ...

Looks like PR #16839 is addressing *both*, @carlescufi can you confirm?

carlescufi, 2019-06-17:
@marc-hb it does indeed. I will flag it accordingly. Thank you for bringing this old issue back to my attention!

closed 2019-06-17 by andrewboie

marc-hb, 2019-07-06:
> > Looks like PR #16839 is addressing both, @carlescufi can you confirm?

> @marc-hb it does indeed.

While looking at https://docs.zephyrproject.org/latest/application/index.html?highlight=cmake#build-an-application I noticed the first examples seen by the reader still don't use -B, I think it's because they explicitly request `:cd-into:`. @carlescufi can you explain why?

PS: found this interesting and relevant discussion https://www.mail-archive.com/cmake-developers@cmake.org/msg16705.html

PR #16839 (2019-06-17): Document building with west by default 
This PR intends to make `west build` a first-class citizen in our documentation. This involves rewriting parts of the zephyr app extension so that we make use of modern CMake features and we do not rely on environment variables anymore, cleaning up the documentation that still refers to `cmake` as an executable directly and finally turn on the option to make west always present in build instructions.
 
See commit messages for additional details.

Fixes #15315