zephyrproject-rtos/zephyr #15316: printk causing usage fault

Pathak94, 2019-04-10:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

ioannisg, 2019-04-10:
Could you give some more details, e.g. which sample/test your are attempting to execute, where do place the code snippet you posted, etc.?

andrewboie, 2019-04-12:
are you using minimal libc or newlib?
what platform are you running this on?
printk doesn't support floating point at all, I can tell you that

closed 2019-05-13 by Pathak94