zephyrproject-rtos/zephyr #15324: Error undefined reference to `__aeabi_uldivmod' when build Zephyr for nrf52_pca10040 board

nhauyeneducation, 2019-04-10:
**Describe the bug**
I try to build zephyr-zephyr-v1.9.2 with zephyr sdk and I meet a issue

  AR      libzephyr.a
  AR      libapplication.a
  LINK    zephyr.lnk
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: drivers/built-in.o: in function `gpio_nrf5_config':
/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/drivers/gpio/gpio_nrf5.c:218: undefined reference to `__popcountsi2'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: subsys/built-in.o: in function `isr_rx_adv':
/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1112: undefined reference to `__aeabi_uldivmod'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1162: undefined reference to `__aeabi_uldivmod'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1163: undefined reference to `__aeabi_uldivmod'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1161: undefined reference to `__aeabi_uldivmod'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: subsys/built-in.o: in function `isr_rx_scan':
/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1338: undefined reference to `__aeabi_uldivmod'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: subsys/built-in.o:/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/subsys/bluetooth/controller/ll_sw/ctrl.c:1608: more undefined references to `__aeabi_uldivmod' follow
collect2: error: ld returned 1 exit status
/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/Makefile:878: recipe for target 'zephyr_prebuilt.elf' failed
make[2]: *** [zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/samples/hello_world/outdir/nrf52_pca10040'
Makefile:178: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory '/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig'
/home/thientran/IoT/IoTivity/iot_newpart/zephyr_project/zephyr-zephyr-v1.9.2_orig/Makefile.inc:82: recipe for target 'all' failed
make: *** [all] Error 2

**Have you ever seen this issue and how to fix it?Please help me. Thank a lot**

**To Reproduce**
Steps to reproduce the behavior:
1. cd sample/hello_world
2. make BOARD=nrf52_pca10040 CROSS_COMPILE=arm-zephyr-eabi-
3. See error

**Expected behavior**
I want to build a simple zephyr example for nrf52 chipset

**Screenshots or console output**
Above

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain Zephyr SDK install in /opt/zephyr-sdk/

**Additional context**
Add any other context about the problem here.

SebastianBoe, 2019-04-10:
I believe this is due to the use of CROSS_COMPILE, can you confirm this by retrying with ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb instead of CROSS_COMPILE?

See https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded for how to do this.

andrewboie, 2019-04-10:
> zephyr-v1.9.2

This is an old, unsupported version of Zephyr.

nhauyeneducation, 2019-04-11:
@andrewboie I am porting IoTivity to Zephyr+NRF52. The Iotivity requires makefile but the lastest version of Zephyr is just support Cmake. Then I found Zephyr 1.9.2 is lastest version that support makefile.

@SebastianBoe I changed the toolchain to third-party **arm-none-eabi-gcc** that download from [here](https://launchpad.net/gcc-arm-embedded/+download) and the error disappeared. 
Thank your support so much.
But the problem seem at Zephyr toochain, is right?

closed 2019-04-11 by nhauyeneducation

SebastianBoe, 2019-04-11:
I'm not sure where the root cause is. I am no longer familiar with how 1.9.2 works.

nhauyeneducation, 2019-04-11:
I was looking in iotivity-lite/port/zephyr/outdir/nrf52_pca10040/zephyr.map and found these lines
 .data          0x00000000        0x0 /opt/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/armv7e-m/libgcc.a(**_aeabi_uldivmod.o**)
 .bss           0x00000000        0x0 /opt/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/armv7e-m/libgcc.a(**_aeabi_uldivmod.o**)

It mean that the __aeabi_uldivmod function implement in the libgcc.a library. I think that the Makefile have to add -lgcc explicitly.

SebastianBoe, 2019-04-11:
It could be, but this is not supported, so it would have to be maintained by the user.